YOODS/rovi #2: NodejsでServiceコールが同期できない

KazukiHiraizumi, 2018-03-07:
【もともとやりたいこと】
RoVIのROS Serviceは、
クライアントから rosservice call /rovi/〇 の形でサービスの実行依頼が来るので、
例えば
  ros.initNode('/rovi/wantblock').then((rosNode)=>
  {
    rosNode.advertiseService('/rovi/somesrv', std_srvs.Trigger, doSome);
  }
  );
という形で、doSome関数内で処理を行うことになる。

ここで
function doSome(req, res) {
　// 下位のサービスを読んだりハードウェアを何かしたりして、
　// それらの最終的な結果が
　// 　　成功なら res.success=true, res.message='成功'
　// 　　失敗なら res.success=false, res.message='失敗'
　return (res.success == true);
}
というcallback関数。

◆問題点◆
　// 下位のサービスを読んだりハードウェアを何かしたりして、
の直列処理を async/await で制御して、成功失敗の待ち合わせ自体は出来ている。
（await Promise.all()で並列処理の待ち合わせもOK。)

ただ、callback関数の中でawaitすると、
await以降がキューに積まれて後で処理されるわけだが、
callback関数自体はawait時点でreturnしてしまう。
（…await以降にresを変更したりreturnしたりの処理を書いても後で処理される…）。
（＝クライアントは結果をcallbackのreturnで得るわけだから、
　　await時点の結果しかクライアントはわからない！）

★やりたいこと★
await時点ではなく、await以降も処理が終わってから
（＝// 下位のサービスを読んだりハードウェアを何かしたりして、
　　を同期化して）
returnできないか？
　　※JavaScriptの仕様上、無理な気がするが。。

無理なら、serviceのreturnは現状のままで、
クライアントには別途結果をparamやtopicで取ってもらうしかない？

▲参考ソース▲
wantblock1.js
wantblock2.js
wantblock3.js

	roscore
	
	rosrun rovi wantblock1.js
	か
	rosrun rovi wantblock2.js
	か
	rosrun rovi wantblock3.js

	rosservice call /rovi/somesrv

	で、問題が見える。

	やりたいことは、
	take@ubuntu:‾/catkin_ws/src/rovi/script$ rosrun rovi wantblock3.js 
	[INFO] [1520410818.903] (ros): Connected to master at http://localhost:11311!
	状態で
	rosservice call /rovi/somesrv
	を打ったら

	[INFO] [1520410821.948] (ros): doSome start.
	[INFO] [1520410826.954] (ros): wake up!
	[INFO] [1520410826.957] (ros): doSome end. now return success=true

	の待ち合わせはできているが、

	この
	doSome end. now return success=true
	がawait以降のキュー処理なので、クライアントには返らず、

	実際にクライアントに返るのは
	rosservice call /rovi/somesrv
	を打った直後の
	take@ubuntu:‾$ rosservice call /rovi/somesrv
	success: False
	message: before await

	であるため、、

	待ち合わせ後に↓を出せないか？ということ。
	take@ubuntu:‾$ rosservice call /rovi/somesrv
	success: True
	message: after await

KazukiHiraizumi, 2018-03-07:
Nodejsなら、非同期処理を強引に同期にする、"Fiber"言うパッケージがあるので、callの同期版callSyncみたいなのを作る、いうことでどうでしょう？
ちなみにBrowserではFoxしかできないようです

KazukiHiraizumi, 2018-03-09:
fs::readSyncとかがどんな実装か見ようとしましたが、これらは組み込み関数ですねえ。jsでは書かれていない。
というとはcallSyncを作るには、C++のros::ServiceClient.callをJSでWrapすることになる。
Cだと型が厳密になってしまうので、汎用的に書けない？どうでしょ？

KazukiHiraizumi, 2018-03-09:
rosnodejsのserviceClient.callはメッセージを(de)serializeとかいう仕組みでストリーム化していますね。callの実体は関数ではなくIPCなので、I/Oレベルではメッセージ型に依存しないいうことか。。。
なかなか巧みの技ですね。

KazukiHiraizumi, 2018-03-09:
ここの仕組みはこのまま置いといて、User向けのサービスコールの仕様を変更してはどうでしょう？
以下のガイドラインです。
1)callの戻り値は、call時点にて処理が受け付けられたかどうかを返す
2)callのResponseで返すものと同じものをTopicにも出す(コールバックでもSubscribeでも同様に使える)。

KazukiHiraizumi, 2018-03-11:
npmでインストールしたのとGitのコードが違いますね。
npmがアップデートされていない。Gitからcloneしたものを上書きすると、以下のようにPromiseを返せば同期できます。
天才的な作りです。すばらしい。
~~~
rosNode.advertiseService('/rovi/projector/run',std_srvs.SetBool, (req,res)=>{
	res.message='call';
	return new Promise(function(resolve){
		setTimeout(function(){
			res.message='back';
			resolve(true);
		},2000);
	});
});
~~~

take-w, 2018-03-13:
すいません、いま（2 days agoの）commentに気付きました。。

> npmでインストールしたのとGitのコードが違いますね。npmがアップデートされていない。
これに気付いてなくて、npmでインストールしたソースを書き換えて同期は実現できました。

ServiceServer.jsの179行目と186行目に↓asyn/awaitを付けるだけ。
    value: async function _handleMessage(client, data) {
      var success = await this._requestCallback(req, resp);

でも、YOODS側でGitでForkしても
https://stackoverflow.com/questions/17509669/how-to-install-an-npm-package-from-github-directly
You can't install any npm package you want from its GitHub source, unless the repo includes a dist folder, and most don't. 
みたいにソース管理がややこしいので、

最新オリジナルGitソースと、教えていただいた「Promiseを返すだけで同期できます」パターンで対応します。

KazukiHiraizumi, 2018-03-13:
ユーザ側にあるだろうパッケージが最新のものになるようにしないとハマりそうです。
Authorに「npmの方がアップデートされてないです」いう連絡しますか？

take-w, 2018-03-13:
> 最新オリジナルGitソースと、教えていただいた「Promiseを返すだけで同期できます」パターンで対応します。

最新オリジナルGitソースにするだけで、現状のRoVIで同期できました。
（結局Promiseを返すのを見やすくするのがasync/await。）

つまりRoVIは現状のまま、rosnodejsは以下のいずれかにすれば動く。

＜A＞
npm install rosnodejsして
ServiceServer.jsの182行目と189行目に↓asyn/awaitを付けるだけ。
value: async function _handleMessage(client, data) {
var success = await this._requestCallback(req, resp);

＜B＞
例えば以下のようにして最新Gitソースを使う
	cd ~
	git clone https://github.com/RethinkRobotics-opensource/rosnodejs

	rpm uninstall rosnodejs
	rpm install rosnodejs
	cd ~/node_modules/rosnodejs
	rm -rf dist
	cp -a ~/rosnodejs/src/ dist

take-w, 2018-03-13:
> というより「Gitにあるのを使えるように」する方が正解かな
どういうわけか
https://stackoverflow.com/questions/17509669/how-to-install-an-npm-package-from-github-directly
You can't install any npm package you want from its GitHub source, unless the repo includes a dist folder, and most don't.
で、rosnodejsも most don't の一人なので、直接Gitのを使うのは難しそうな。。

Forkしてdist/（実際はsrc/）をGitに登録、というのもなんか汚い。

ひとまず上記＜A＞か＜B＞をインストール手順書的なものに書いておけばいいかと思います。
（そのうち、Authorがテストなど終わったら最新Gitをnpmに入れてくれるはずなので、それまでひとまずの対応。）

take-w, 2018-03-13:
Dropbox\NEDO\設計図書\クライアントからのRosService結果取得について\rosnodejsのadvertiseServiceのcallback関数を同期化する方法はないか.txt

に上記をまとめて追記しておきました。

KazukiHiraizumi, 2018-03-16:
YOODSでは、async、awaitを使った記述に統一します。
↓がわかりやすいように思いました
https://javascript.info/async-await

closed 2019-02-16 by KazukiHiraizumi