YOODS/rovi #6: camera_aravisソフトウェアトリガの失敗

KazukiHiraizumi, 2018-03-15:
ソフトウェアトリガのQC-1/長時間運転試験を行っていますが、かなりの頻度でトリガがかからないことがある。監視タイマでリトライはしても、タイマーをあまり短くできない(トリガの２度入りが起きる)。
ハードウェアトリガなら安定している。時々インタバルが長い時(200〜300ミリ秒程度)があるものの、カメラの内部的なインタロックがある？ためか長時間でも問題がおきない。
結論的には、ライブ＝ハードトリガ、ということになる。

KazukiHiraizumi, 2018-03-15:
FrameRateの設定基準はどうなっているのかな？
お客さんのプラットフォームは様々だろうから、診断機能を組み込む必要あり。

KazukiHiraizumi, 2018-03-15:
TriggerModeをOffからOnに変えた時、FrameRateは1000に書き変わる。これはcamera_aravisのソースを直す必要あり

take-w, 2018-03-19:
2018/3/16(金)の打ち合わせでの
　TriggerMode=Offにして、ストリーミングモードにする。
　しかし1FPSくらいでしか動かないのは？
　camera_l, camera_rではなく、片側（camera_l）だけにして、動かしてみる。

⇒
片側でも、安定して動くのは
（＝Frame error: ARV_BUFFER_STATUS_TIMEOUT
　　エラーが出ないのは）
FPS 2.0以下くらい。
（FPS 2.125や2.1なら基本大丈夫だが、たまにエラーが塊になって出る。）

KazukiHiraizumi, 2018-03-19:
それは大問題ですね。
VMではないLinux環境でやってみましょうか？
ありますか？

take-w, 2018-03-19:
ないです。。

3/23(金)に原田さんがセットアップ済み予定のJetsonとつないでやってみるのがいいでしょうか。

KazukiHiraizumi, 2018-03-19:
CelelonですがNativeのLinuxPCを持っているので、次回DRで動かしてみましょう。
他のところをすすめてください

KazukiHiraizumi, 2018-03-19:
@take-w 
解像度を下げたらどうなるかみてもらえますか？私の方はVGA(640x480)のカメラですが、これだと上のPCにaravisとimage_viewのみで50FPSくらいまでいけます。
解像度の変更はrosparamではできません。arv-toolいうのでやります。
~~~
arv-tool-0.4 control Width=640 Height=480
~~~

take-w, 2018-03-19:
（これまでは1280x1024だったのを）
640x480にしたところ、
aravis+image_viewで、

片側だと、8FPSくらいならOK。
（8.5や9だとエラーが少し出始める。10だと明らかにエラーが出てくる。）

両側だと、4FPS & 4FPS くらいならOK。
（何もしていない状態なら6, 6くらいならOKだが、image_viewとかGUIクリックとかで少しエラーが出る。）

でした。

画素数が1/4になったからFPSも4倍まで行けるようになった、という感じでしょうか。🤔

320x240にすると、片側32, 両側16&16、とまではいかずに、OKなのはもう少し低い値になりました。
32x32だと、両側100&100でもOK。
しかし320x240でも視野が狭いし、32x32はどこが映っているのかわからない視野の狭さ。
（RoVIでどのくらいの視野の広さが必要とされているかはわかりませんが。。）

KazukiHiraizumi, 2018-03-19:
@take-w 
いいIssueがありました
https://answers.ros.org/question/132446/aravis-camnode-frame-error-arv_buffer_status_timeout/
自分のPCでやってみましたが、いままで１００FPSでも動くようになりました。
mtuの設定はカメラが記憶しているようなので注意ください。
paramのcamera/mtuにもあるのですが、これがあまり効いていないので、カメラに直接した方がいいです。初期値は1500なのでこれを9000にします
~~~
arv-tool-0.4 control GevSCPSPacketSize=9000
~~~
これでcamnodeを起動すると起動メッセージの最後あたりに、paramのmtuとは無関係に
~~~
[ INFO] [1521499733.508251432]:     Network mtu          = 9000
~~~
と出ます。
PC側のmtu設定は、私の古いPCだと1500が上限だったので、こっちも注意ください。

take-w, 2018-03-20:
ありがとうございます。

そもそもGevSCPSPacketSizeの初期値が576だったので、1500にしたら少しはFPSが改善しました。
（せいぜい2倍ちょっとですが。。）

次に9000ですが、VMware Workstation Playerの限界らしく、1500より上にすると常に
Frame error: ARV_BUFFER_STATUS_TIMEOUT
でだめでした。
（ホストWindows側をJumbo Packet 9014Byteにして、Ubuntuでmtu 9000などしてもだめ。
　有料VMware vSphereだとJumbo Packetがいけるらしい。。）

詰まっている場所はわかったので、ひとまずよしとして。

開発用に、VMwareではない、ちゃんとしたUbuntu PCを用意してもらう方がいいかも・・？

KazukiHiraizumi, 2018-03-20:
UbuntuよりLinuxMintがいいですよ。
https://linuxmint.com/
Ubuntu起源なので互換だし、普段PCもこれで済む。

Kan-Harada, 2018-03-20:
Jetsonを使えば良い気がしますが・・・どうでしょうか?

Kan-Harada, 2018-03-20:
MTU9000の設定はGigE Visionでは必須と聞いています。ただ、以前YJC-3でやろうとしたらできなかった。これは使っているPHY等に依存するのでは・・というところで終わっていました。

take-w, 2018-03-20:
> 普段PCもこれで済む。
VPNがつながらない？とかいう話もありますし、完全にWindowsから移行してこれ一台、はさすがに無理かなと。

でもNEDO開発用LinuxマシンのOSとしてはMintでいいかもですね。

take-w, 2018-03-20:
> Jetsonを使えば良い気が

Jetsonだと開発環境と動作環境がわかれてまた話がややこしくなるような。。

take-w, 2018-03-20:
> MTU9000の設定はGigE Visionでは必須と聞いています。
でも、WindowsのeBUS Playerだと、MTUはいじらずに（Jumbo Packet Disabledで）60fps行けるわけでして。

UbuntuのeBUS Playerも60fpsまではいけないがそこそこいけるわけで、、？

camera_aravisのドライバとの絡みが？（何かよくわからなくなってきました。）

Kan-Harada, 2018-03-20:
@take-w ノートPCならばいくつかありますがハードディスクを載せ換えて使いますか?

take-w, 2018-03-20:
はい。適当なノートPCで。ハードディスクも適当で。
（UbuntuかMintを入れてネイティブに使います。）

まっさらハードディスクに乗せ換えて、Ubuntu or Mintを入れるって感じでしょうか。

take-w, 2018-03-26:
2018/3/23(金)のFRONTIERノートPCにUbuntu 16, ROSなど入れて試してみました。

デフォルト状態
	PC側のmtuは1500
	arv-tool-0.4 control GevSCPSPacketSize は 576
	arv-tool-0.4 control Width Height は 1280x1024
だと、
カメラ1台でも2台でも、FPS 40くらいまでOK.
（カメラ1台でも2台でも、FPS 45だとかなりエラー。）

MTU 9000状態
	PC側のmtuは9000
	arv-tool-0.4 control GevSCPSPacketSize は 9000
	arv-tool-0.4 control Width Height は 1280x1024
だと、
カメラ2台でFPS 30, 30なら安定する。（60, 60とかはエラー多発。）

MTU 9000で解像度を下げると
	PC側のmtuは9000
	arv-tool-0.4 control GevSCPSPacketSize は 9000
	arv-tool-0.4 control Width Height は 640x480
カメラ2台でFPS 1000, 1000でも安定する。

よって、（MTUは9000にするとして、）解像度とFPSの関係としては

＜A＞
解像度：1280x1024
FPS: 30, 30

＜B＞
解像度：640x480
FPS: 1000, 1000

とかなります。

ただ、別環境ではBでも1000, 1000とか出せるとは限らないので、
結局Aを基本に考えておくのがいいでしょうか？

take-w, 2018-03-26:
あと、
	arv-tool-0.4 control GevSCPSPacketSize は 9000
	arv-tool-0.4 control Width Height は 1280x1024
にするのは、YCAM側を再起動すると初期値に戻ってしまって、これらの値になる保証がないので、
RoVIの起動時にYCAMへコマンドを打ち込む必要がありそうです。

KazukiHiraizumi, 2018-03-26:
pshift_genpc.jsでライブと13枚撮影まで実装しました。
Chatの方にコメント入れました。
カメラ２台で動かすときは、ycam1h.jsのコメントを外してください。
プロジェクタの制御はYCAMがないとテストが出来ないので、水曜日にやります。

take-w, 2018-03-26:
> 私の手元にあるBaslerのカメラなら、Paramerのｍｔuがcamnode起動時に設定されますね。
こちらのFRONTIER PCとYCAMの組み合わせでも同様にParameterの値がarv-toolへ設定反映されました。
なので
> RoVIの起動時にYCAMへコマンドを打ち込む必要がありそうです。
は無し、で。

> PC側は1500のままでもOKそうです。送信時しか意味無いのか？
これも確認できました。
なのでPC側は特にmtuを変えるようにはGettingStarted.mdには書かないことにします。

take-w, 2018-03-26:
あぁでも、RoVIは起動しっぱなしで、YCAMの電源をOFF->ONすると、mtuは576とかの初期値に戻りますね。

そういうのも異常監視で見ておいて、Parameterを反映させればいいか？

KazukiHiraizumi, 2018-03-27:
camnodeが落ちたらycam1h.jsで監視して再起動します。

take-w, 2018-03-27:
> Parameterの値がarv-toolへ設定反映され
は、launchファイル内で
<rosparam command="load" ns="/rovi" file="$(find rovi)/yaml/rovi_rosparam_dump.yaml" />
のようにしてから
<node>で各種ノード（/rovi以下のParameterを使っている）を起動しているからですね。

ycam1h.jsの中からの起動でも同様にrosparam loadをすれば再設定されると思います。
（まぁ細かいことは明日。）

KazukiHiraizumi, 2019-02-16:
最終的にはライブモードもソフトウェアでプロジェクタからトリガ（ワンショット）を出すようになりました

closed 2019-02-16 by KazukiHiraizumi