YOODS/rovi #8: 共通サービスのインタフェース

KazukiHiraizumi, 2018-03-17:
RoViパッケージのノードで、以下の機能が必要であれば、共通のインタフェースに揃えたいと思います。
1. パラメータのリロード
2. 汎用コマンド
## パラメータのリロード reload
サービス名は.../reloadとします。
サービス型はROSで定義済のstd_srvs::Trigger型とします。TriggerはRequestメッセージがないので、呼び出しは下のようになります。
~~~
rosservice call .../reload
~~~
## 汎用コマンド parse
特に処理速度が重要でないサービスにも、いちいち型定義するのは面倒なので、文字列を投げてやってもらう的サービスを規定したいと思います。
サービス名は.../parseとします。
Syntaxは下のようにします。
~~~
rosservice call .../parse <i>cmd [{key:value}]</i>
~~~

KazukiHiraizumi, 2018-03-17:
parse続き...
1. .../parseに"get"というコマンドを投げます。"get"コマンドは引数を必要としない(あるいは省略可能)とすると以下の入力となります。
~~~
rosservice call .../parse get
~~~
2. .../parseに"set"という、引数が必要なコマンドを投げます。
~~~
rosservice call .../parse 'set {"Gain":10,"TriggerMode":"On"}'
~~~
引数はJSON形式で記述します。またコマンドと引数は(')で括り、keyは(")で括ります。

KazukiHiraizumi, 2018-03-17:
parse続き...
parseのサーバサイドの実装例です。
~~~
rosNode.advertiseService('..../parse',rovi_srvs.dialog, (req,res)=>{
	let cmd=req.hello;
	let lbk=cmd.indexOf('{');
	let obj={};
	if(lbk>0){
		cmd=req.hello.substring(0,lbk).trim();
		obj=JSON.parse(req.hello.substring(lbk));
	}
	ros.log.info('parsed:'+cmd+' arg:'+JSON.stringify(obj));
	switch(cmd){
	case 'get':
		return new Promise((resolve)=>{
			res.answer='{Gain:100}';
			resolve(true);
		});
	case 'set':
		return Promise.resolve(true);
	}
});
~~~

take-w, 2018-03-19:
質問です。

parseサービスは、現状はcmdとしてget/setのみ、ということは、
parse getは ROS Parameterのget
parse setは ROS Parameterのset
を行うということですよね？

例えば左側カメラのゲインを設定したい場合は
rosservice call .../parse 'set {"/rovi/cam_l/camera/Gain":10}'
という理解でよいでしょうか？
（keyはROS Parameter名そのもの。）

そして、3DのためのGain設定なら、
rosservice call .../parse 'set {"Gain":10}'
とする。
（前述の左側カメラのゲインパターンとは違い、keyは絶対パス形式ではない。
　はっきりと「"RoVI-3D-Gain"」みたいなkey文字列がよいかもしれない。）

KazukiHiraizumi, 2018-03-20:
.../parse "string" という書式までで、stringのとこに来る、具体的なコマンドまでは規定してないです。
ノード側の実装に委ねるいうことで考えています。
ただJSなら実装も簡単なので、パラメータ設定のコマンド(set,get)だけはルール化してもよいかもしれません。
いまのところパラメータの設定＆反映のルールは
１)dynamic_reconfigure対応のノード(たぶん既存のノード)であれば、dynparam
2)YOODSで新製するものはrosparamの後に、call .../reload
3)１，２以外はノードを再起動

take-w, 2018-03-20:
> rosservice call /rovi/upper/param_set {"3D-Gain":10, "3D-ExposureTimeAbs":12345}
も形は合わせて（パラメータの絶対パス形式で）
> rosservice call /rovi/upper/param_set {"/rovi/3d/Gain":10, "/rovi/3d/ExposureTimeAbs":12345}
のようにすることにします。

こうすればparam dumpできる。
（実際には/rovi/3d/Gainの値などは必要に応じてカメラに設定に行く、という当初予定通り。）

KazukiHiraizumi, 2019-02-16:
Paramer変更のNotifiierを使う

closed 2019-02-16 by KazukiHiraizumi