YOODS/rovi #10: ドキュメント(*.md)についての話の場

take-w, 2018-03-27:
@KazukiHiraizumi @Kan-Harada 
ドキュメント(*.md)についてコメントできるようにIssueを発行しました。

ひとまず1個だけ↓。

Calibration.md
の最後の方に
「ROSのクセですが、：の後にはスペースが必要なので注意しましょう」
とありますが、
コロンの後ろにスペースが必要なのは（ROSのクセというより）YAMLの仕様なので、
（…ROS ParameterもdynparamもYAML文法で書くようになっている…）

「パラメータ書式はYAML文法にしたがうため、：の後にはスペースが必要なので注意しましょう」
などでいかがでしょうか。

take-w, 2018-03-27:
GettingStarted.mdもとりあえず書いた（実行手順は大枠のみ）ので、チェック＆コメントお願いします。

KazukiHiraizumi, 2018-03-27:
動かせるまでの手順はなかなか多いですね。
1. npmで必須ものはrosnodejsだけでは？
2. RoVI起動のあとに動作確認方法が要りそう
  1. ライブ画像のViewing方法（Topic）
  2. 点群画像のViewing方法
あとはちょっと誰かを実験台にして見てみるのが一番です。

KazukiHiraizumi, 2018-03-27:
さらにそのあと、画像調整みたいなセクションで、GainやExposureの調整が要りますね。
それで最後の最後に(Appendixかな)、
- Parameter一覧
- Topic一覧
- Service一覧
でしょうか

take-w, 2018-03-27:
npm install ws
npm install opencv
npm install canvas
はブラウザでのアレコレのためでしたでしょうか。

となると今回は
npm install ws
npm install canvas
はインストール不要は確定。

ただ、
npm install opencv
については、その中でインストールされる
minimist
js-yaml
というのをjs内で使っています。
（前者は起動引数解釈、後者はYAML文法解釈）

なので、以下のどちらかになります。どうしましょう？

＜A＞
npm install opencv
（上記minimistとjs-yamlのため。将来opencvが必要になった場合に備えても。）

＜B＞
npm install minimist
npm install js-yaml
（はっきり2つ入れる。）

take-w, 2018-03-27:
> 最後の最後に(Appendixかな)、
> Parameter一覧
> Topic一覧
> Service一覧
了解です。（これらをどこに書くか悩んでました。Appendixにします。）

closed 2019-03-23 by KazukiHiraizumi