YOODS/rovi #11: /rovi/pshift_genpcの動作がおかしい

take-w, 2018-03-28:
take@take-NXC154:~$ cat .ros/log/latest/rovi-pshift_genpc-5-stdout.log
[INFO] [1522221397.164] (ros): Connected to master at http://localhost:11311!
[INFO] [1522221397.215] (ros): YPJ connected
projector :ERR//undefined

projector :Dlp.X>
rosrun:stdout:[ INFO] [1522221397.274631464]: Attached cameras:

rosrun:stdout:[ INFO] [1522221397.279049710]: Attached cameras:

rosrun:stdout:[ INFO] [1522221398.301503164]: # Interfaces: 2
[ INFO] [1522221398.301599724]: # Devices: 2

rosrun:stdout:[ INFO] [1522221398.301648068]: Device0: SENTECH-17AB755
[ INFO] [1522221398.301682337]: Device1: SENTECH-17AB756
[ INFO] [1522221398.301729024]: Opening: SENTECH-17AB755

rosrun:stdout:[ INFO] [1522221398.325274683]: # Interfaces: 2

rosrun:stdout:[ INFO] [1522221398.325380174]: # Devices: 2
[ INFO] [1522221398.325425374]: Device0: SENTECH-17AB755
[ INFO] [1522221398.325466358]: Device1: SENTECH-17AB756
[ INFO] [1522221398.325517597]: Opening: SENTECH-17AB756

rosrun:stdout:[ INFO] [1522221398.431927591]: Opened: SENTECH-17AB755

rosrun:stdout:[ INFO] [1522221398.438502368]: Opened: SENTECH-17AB756

rosrun:stdout:[ INFO] [1522221398.500640673]: Read parameter (float) AcquisitionFrameRate: 30.000000

rosrun:stdout:[ INFO] [1522221398.506521264]: Read parameter (string) ExposureAuto: Off

rosrun:stdout:[ INFO] [1522221398.506898389]: Read parameter (float) AcquisitionFrameRate: 30.000000

rosrun:stdout:[ INFO] [1522221398.512381599]: Read parameter (float) ExposureTimeAbs: 2000.000000

rosrun:stdout:[ INFO] [1522221398.512748124]: Read parameter (string) ExposureAuto: Off

rosrun:stdout:[ INFO] [1522221398.518052111]: Read parameter (float) Gain: 10.000000

rosrun:stdout:[ INFO] [1522221398.518513233]: Read parameter (float) ExposureTimeAbs: 2000.000000

rosrun:stdout:[ INFO] [1522221398.523969976]: Read parameter (string) GainAuto: Off

rosrun:stdout:[ INFO] [1522221398.524122145]: Read parameter (float) Gain: 10.000000

rosrun:stdout:[ INFO] [1522221398.529952203]: Read parameter (string) TriggerMode: Off

rosrun:stdout:[ INFO] [1522221398.529986191]: Read parameter (string) GainAuto: Off

rosrun:stdout:[ INFO] [1522221398.530068146]: Read parameter (string) TriggerSource: Line1

rosrun:stderr:[ERROR] [1522221398.532248078]: ros parameter url not set in camnode where's the camera_info file?

rosrun:stdout:[ INFO] [1522221398.535826355]: Read parameter (string) TriggerMode: Off

rosrun:stdout:[ INFO] [1522221398.535931637]: Read parameter (string) TriggerSource: Line1

rosrun:stderr:[ERROR] [1522221398.538861657]: ros parameter url not set in camnode where's the camera_info file?

rosrun:stdout:[ INFO] [1522221398.626156640]: Set gain = 10.000000

rosrun:stdout:[ INFO] [1522221398.630592571]: Set gain = 10.000000

rosrun:stdout:[ INFO] [1522221398.631745303]: Set AcquisitionFrameRate = 30.000000

rosrun:stdout:[ INFO] [1522221398.636197763]: Set AcquisitionFrameRate = 30.000000

rosrun:stdout:[ INFO] [1522221398.637703742]: Camera does not support FocusPos.
[ INFO] [1522221398.637768426]: Set mtu = 9000

rosrun:stdout:[ INFO] [1522221398.642165541]: Camera does not support FocusPos.

rosrun:stdout:[ INFO] [1522221398.642235715]: Set mtu = 9000

rosrun:stdout:[ INFO] [1522221399.638424796]: Get mtu = 9000

rosrun:stdout:[ INFO] [1522221399.642821468]: Get mtu = 9000

rosrun:stdout:[ INFO] [1522221401.693946835]:     Using Camera Configuration:
[ INFO] [1522221401.694045795]:     ---------------------------

rosrun:stdout:[ INFO] [1522221401.694085120]:     Vendor name          = SENTECH
[ INFO] [1522221401.694120937]:     Model name           = STC_BBE132GE
[ INFO] [1522221401.694162131]:     Device id            = 17AB755
[ INFO] [1522221401.694203823]:     Sensor width         = 1280
[ INFO] [1522221401.694239261]:     Sensor height        = 1024
[ INFO] [1522221401.694282220]:     ROI x,y,w,h          = 0, 0, 1280, 1024
[ INFO] [1522221401.694324978]:     Pixel format         = mono8
[ INFO] [1522221401.694364596]:     BytesPerPixel        = 1

rosrun:stdout:[ INFO] [1522221401.694550475]:     Acquisition Mode     = Continuous
[ INFO] [1522221401.694634051]:     Trigger Mode         = Off

rosrun:stdout:[ INFO] [1522221401.696495495]:     Using Camera Configuration:
[ INFO] [1522221401.696600932]:     ---------------------------

rosrun:stdout:[ INFO] [1522221401.696660007]:     Vendor name          = SENTECH
[ INFO] [1522221401.696722073]:     Model name           = STC_BBE132GE
[ INFO] [1522221401.696788037]:     Device id            = 17AB756
[ INFO] [1522221401.696846809]:     Sensor width         = 1280
[ INFO] [1522221401.696899660]:     Sensor height        = 1024
[ INFO] [1522221401.696955181]:     ROI x,y,w,h          = 0, 0, 1280, 1024
[ INFO] [1522221401.697010489]:     Pixel format         = mono8

rosrun:stdout:[ INFO] [1522221401.697063521]:     BytesPerPixel        = 1
[ INFO] [1522221401.697270929]:     Acquisition Mode     = Continuous

rosrun:stdout:[ INFO] [1522221401.697363596]:     Trigger Mode         = Off

rosrun:stdout:[ INFO] [1522221401.698187529]:     Trigger Source       = Software
[ INFO] [1522221401.698257714]:     Can set FrameRate:     True

rosrun:stdout:[ INFO] [1522221401.698308391]:     AcquisitionFrameRate = 30 hz
[ INFO] [1522221401.698348392]:     Can set Exposure:      True
[ INFO] [1522221401.698379717]:     Can set ExposureAuto:  True
[ INFO] [1522221401.698413563]:     Exposure             = 2000 us in range [10,1.00072e+06]
[ INFO] [1522221401.698444644]:     Can set Gain:          True
[ INFO] [1522221401.698469886]:     Can set GainAuto:      True
[ INFO] [1522221401.698508170]:     Gain                 = 10.000000 % in range [0.000000,63.000000]
[ INFO] [1522221401.698538792]:     Can set FocusPos:      False

rosrun:stdout:[ INFO] [1522221401.698605315]:     Network mtu          = 9000
[ INFO] [1522221401.698662160]:     ---------------------------

rosrun:stdout:[ INFO] [1522221401.700957271]:     Trigger Source       = Software
[ INFO] [1522221401.701037297]:     Can set FrameRate:     True

rosrun:stdout:[ INFO] [1522221401.701098529]:     AcquisitionFrameRate = 30 hz
[ INFO] [1522221401.701158074]:     Can set Exposure:      True
[ INFO] [1522221401.701221776]:     Can set ExposureAuto:  True
[ INFO] [1522221401.701298037]:     Exposure             = 2000 us in range [10,1.00072e+06]

rosrun:stdout:[ INFO] [1522221401.701360734]:     Can set Gain:          True
[ INFO] [1522221401.701419856]:     Can set GainAuto:      True
[ INFO] [1522221401.701490669]:     Gain                 = 10.000000 % in range [0.000000,63.000000]

rosrun:stdout:[ INFO] [1522221401.701550525]:     Can set FocusPos:      False
[ INFO] [1522221401.701627722]:     Network mtu          = 9000
[ INFO] [1522221401.701684697]:     ---------------------------

rosrun:stdout:[ INFO] [1522221401.799464321]: using default calibration URL

rosrun:stdout:[ INFO] [1522221401.799554143]: camera calibration URL: file:///home/take/.ros/camera_info/17AB756.yaml
[ INFO] [1522221401.799625637]: Unable to open camera calibration file [/home/take/.ros/camera_info/17AB756.yaml]
[ WARN] [1522221401.799653697]: Camera calibration file /home/take/.ros/camera_info/17AB756.yaml not found.

rosrun:stdout:[ INFO] [1522221401.877057926]: using default calibration URL

rosrun:stdout:[ INFO] [1522221401.877139541]: camera calibration URL: file:///home/take/.ros/camera_info/17AB755.yaml
[ INFO] [1522221401.877277466]: Unable to open camera calibration file [/home/take/.ros/camera_info/17AB755.yaml]
[ WARN] [1522221401.877334699]: Camera calibration file /home/take/.ros/camera_info/17AB755.yaml not found.

projector ://cmd:start stop 2

projector ://cmd:start stop 0
projector :
cmd:p2=1:OK

projector :Dlp.X>
projector ://cmd:start stop 2

projector ://cmd:start stop 0
projector :
cmd:p2=1:OK
projector :

projector :Dlp.X>
rosrun:stdout:[ INFO] [1522221694.171595834]: Set TriggerMode = On

projector ://cmd:set exposure

projector ://exposure: 30720
projector :

projector :cmd:x30=1:OK
projector :

projector :Dlp.X>cmd:p50=1:OK
projector :

projector :Dlp.X>//test pattern start : 0
projector :

projector ://cmd:start stop 2
projector :

rosrun:stdout:[ INFO] [1522221696.196057170]: Set TriggerMode = Off

projector ://cmd:start stop 0
projector :

projector ://test pattern end
projector :

projector :cmd:pffffff=1:OK
Dlp.X>//cmd:start stop 2

projector ://cmd:start stop 0

projector :cmd:p2=1:OK
projector :

projector :Dlp.X>
take@take-NXC154:~$

take-w, 2018-03-28:
pffffff部分が悪かったということでiffffffに修正。

take-w, 2018-03-28:
新問題。

take@take-NXC154:~/catkin_ws/src/rovi$ rosservice call /rovi/pshift_genpc
ERROR: service [/rovi/pshift_genpc] responded with an error: Unable to handle service call

のようになる。

↓
yaml/rovi_param.yaml
で
pshift_genpc:
  projector:
    Interval: ○
を50から100へ、で修正。

take-w, 2018-03-28:
新問題2。

カメラなど起動後、初回だけ、必ず
take@take-NXC154:~/catkin_ws/src/rovi$ rosservice call /rovi/pshift_genpc
ERROR: service [/rovi/pshift_genpc] responded with an error: Unable to handle service call
となる。
この場合、発光もしない。

take-w, 2018-03-28:
新問題3。

 rosservice call /rovi/pshift_genpc
でTrueが返ってきた後、

rosrun image_view image_view image:=/rovi/cam_l/view
rosservice call /rovi/pshift_genpc/parse 'view 0'
rosservice call /rovi/pshift_genpc/parse 'view 1'
などしても、rect画像がview *ですべて同じ。
（撮影最初の1枚目が常に出る？）

ライブで/rovi/cam_l/camera/image_raw, /rovi/cam_l/image_rectをimage_viewで出していて、
pshift_genpcをcallすると、
本来はその間image_rectは更新されないはずが、更新されてしまっている、というのが何かおかしい。

KazukiHiraizumi, 2018-03-28:
@take-w 
pshift_genpc.js修正しました。私の方から先にpushしていいのかな？

take-w, 2018-03-28:
どんどんpushしていただいてOKです。😙

KazukiHiraizumi, 2018-03-28:
pushしました。昨日の変更とmergeしてみてください

take-w, 2018-03-29:
最新ソースで実行したところ、
ライブでcamera/image_rawとimage_rectを見ている状態で
pshift_genpcを呼ぶと、
確かにその間はimage_rect側は止まっているようです。

ただ、サービス自体は↓失敗しています。
ERROR: service [/rovi/pshift_genpc] responded with an error: Unable to handle service call

また、/rovi/cam_l/viewをimage_viewで出してparse view 0などしても、画像は何も出てこず、
No image captured: TypeError: cannot read property 'length' of undefined
というWARNが出ます。
（ソースにログを入れて確認すると、case 'view':でのviewOut()呼び出し内で、capt_Lもcapt_Rもundefinedになっているのが原因の模様。）
（結局、/rovi/pshift_genpcサービス自体がうまくいっていないからcapt_L, capt_Rにも正しく入っていないと？）

KazukiHiraizumi, 2018-03-29:
TriggerがLine1から来ないのか。
サービスの最初の方でTriggerModeをOnにセットしてる行をコメントアウトしてやってみてもらえますか？

take-w, 2018-03-29:
コメントアウトして実行しても現象は変わりませんでした。

現状の問題点をまとめると：
A. 初回のpshift_genpc callで発光しない。（呼び出しERROR）
B. 2回目以降のpshift_genpc callでは発光するが呼び出しERRORで、parse view Nでcapt_Lやcapt_Rがundefinedとなる。
C. launchファイル終了時、なぜか発光する。
D. なんか先ほど、やたらと発光がゆっくりと長く起こった。（再現せず。）（パターン間の間隔が100msecより大きい感じで、全体で13パターンどころか30パターン以上は光っていた。）

KazukiHiraizumi, 2018-03-29:
何かおかしいですね。
明日もう一度来れますか？

take-w, 2018-03-29:
山口に行くのはOKです。開始時間は指定してください。

（できればこちらで今日ソースを修正してしまえればベストですが。）

KazukiHiraizumi, 2018-03-29:
１５時でいいですか？
それまでに動くようになれば中止しましょう

take-w, 2018-03-29:
了解です。

KazukiHiraizumi, 2018-03-29:
Issueの最初のログですが、
1. 起動時のログ
2. pshift...をCallした時のログ
に分けるとどうなりますか？
というか現状どうなのでしょうか？

take-w, 2018-03-29:
[NotDo-TriggerModeOn.txt](https://github.com/YOODS/rovi/files/1860217/NotDo-TriggerModeOn.txt)
[DO-TriggerModeOn.txt](https://github.com/YOODS/rovi/files/1860218/DO-TriggerModeOn.txt)

take-w, 2018-03-29:
↑ひとまずさきほどのpushについてのログと説明です。

けど、このログはlaunchのものなので、最初のログ（latest/*.log）はこれから取ります。

take-w, 2018-03-29:
発光処理
（タイミングとか、初回発光せず、launch終了に発光する、とか）
がおかしいのかな？と思えますが。。

take-w, 2018-03-29:
DO-TriggerModeOn.txt
内に書き忘れましたが、こちらはService callがERRORです。

take-w, 2018-03-29:
[NotDo-TMOn_latest.txt](https://github.com/YOODS/rovi/files/1860261/NotDo-TMOn_latest.txt)
[DO-TMOn_latest.txt](https://github.com/YOODS/rovi/files/1860262/DO-TMOn_latest.txt)
最初相当のログです。

KazukiHiraizumi, 2018-03-29:
画像のキャプチャ間隔が異様に短いですね。なんでやろ？

KazukiHiraizumi, 2018-03-29:
左右の画像が交互に来ているものと思い込んでいましたが、ライブでも実は違うのかもしれない気がしてきました。

KazukiHiraizumi, 2018-03-29:
そもそもUDPはEtherCatみたいなデバイスに均等に権利が回ってくるプロトコルではないですよね。。。

take-w, 2018-03-29:
あぁ、UDPなんですか。

その辺含め、YCAMのプロジェクターインターフェースがどうなっているのかほぼ全く理解してませんけども (恥)

今日の解決は無理っぽいですね。

Kan-Harada, 2018-03-29:
ソースを見ていないのですが、一般的に2台のGigE Visionのカメラが同時にシャッターを切るので
パケットは両方のカメラから非同期にやってきます。

現状のカメラのトリガ部分は平泉さんが作られたもので、

1. PICマイコンから左右のカメラに同時にシャッタートリガをハードウェア的に出す
2. 設定された露光時間の後、全画素データを順次UDPパケットで送出
3. 左右のカメラからのUDPパケットは完全に非同期で、その調停はHUBが行っている

という形になります。このHUB任せの部分をなんとかしたく、YCAM3D-IIIでは左右のカメラからのピクセルデータを一旦FPGAでまとめて、1枚の画像として送出するようにしました。(まだ動いていませんが)

KazukiHiraizumi, 2018-03-29:
TriggerをOnにすると左右の撮影タイミングが同期してしまうので、より変なことになってそう。
今日はこの辺にしましょう。
Win版は動いているのが不思議

KazukiHiraizumi, 2018-03-29:
このあたりの話でしょうか
https://www.baslerweb.com/jp/sales-support/downloads/document-downloads/controlling-packet-transmission-timing-with-delays/

KazukiHiraizumi, 2018-03-29:
上のBaslerのドキュメントに書かれている複数カメラ使用時の注意点そのものかもですね。
本日のテストとしてはTriggerModeはOnにて
1. カメラ１個にして１３枚の画像が所定のインターバルで取得できるかどうか。
2. 左右カメラのタイミング調整をして同様の状態になるかどうか

1.については、ハード的にトリガのラインを外す。ソフトならRevisionを戻してycam1h.jsの右カメラを無効にしてもできるが・・・

KazukiHiraizumi, 2018-03-30:
TriggerDelayAbsというパラメータがあるので、これでBaslerのドキュメントに従って左右のカメラの送信タイミングをズラす、いう方針です。
@take-w 
今日はこっち来ますか？

KazukiHiraizumi, 2018-03-30:
カメラ１台あたりのGigEの転送時間は
1.3Mbytes x 0.008 μs = 0.0104sec
だから、今の30FPSいう設定はかなり一杯ですね。
ライブスタート(TriggerMode=Off)を単に10msecくらいズラせばいいだけなのか？このあたりはよく見ておきましょう。

take-w, 2018-03-30:
> 今日はこっち来ますか？

はい。15時開始で行きます。

KazukiHiraizumi, 2018-03-30:
> カメラ１台あたりのGigEの転送時間は
> 1.3Mbytes x 0.008 μs = 0.0104sec
> だから、今の30FPSいう設定はかなり一杯ですね。
> ライブスタート(TriggerMode=Off)を単に10msecくらいズラせばいいだけなのか？このあたりはよく見ておきましょう。
これはGigE回線の専有時間だから、センサー⇒GigEコントローラの転送時間はもっと長い。Baslerの例だとそっちは、この４倍程度なので0.04sec程度が１カメラあたりのターンアラウンド(GigE＋センサー)。
となると２台で、フレーム単位で衝突が起こらないようにするには、１０FPSが妥当なとこ。
外部Triggerであればパケット単位でズラせるが、ライブはそこまで拘らないでいいかな.。

closed 2019-03-23 by KazukiHiraizumi