YOODS/rovi #12: camera_aravisを複数起動するとおかしい

KazukiHiraizumi, 2018-04-01:
camera_aravisが複数

KazukiHiraizumi, 2018-04-01:
stdin仕様
- GigE Visionのパラメータをそのまま渡す
  - パラメータ名 値 \n

stdout仕様
- JSON形式
  - {"shm":id}   // id: SysV共有メモリID
  - {"capt":offset}   // offset: 画像転送先メモリオフセット

KazukiHiraizumi, 2018-04-02:
basler_example/grabberのテスト例  
最初に共有メモリIDが出力される。とりあえず初期はストリーミングOFFなので、ここで入力待ち
~~~
$ ./grabber
{"shm":279314488} 
~~~
ストリーミングONにするにはstdinに以下入力
~~~
TriggerMode Off
~~~
画像メモリ上に用意できたらstdoutでNotifyする
~~~
{"capt":307200}
{"capt":0}
{"capt":307200}
{"capt":0}
{"capt":307200}
{"capt":0}
{"capt":307200}
{"capt":0}
{"capt":307200}
~~~
左右カメラから交互に画像が来ればOKとしよう。   　
FrameRateの変更などもストリーミング中でも可能。ただしGEVの素のパラメータで入力する(camera_aravisは内部で変換しているものもあったりで一貫性がない)
~~~
AcquisitionFrameRateAbs 20
~~~

KazukiHiraizumi, 2018-04-02:
ROSでのテスト

画像が正しいかどうかのテストは、共有メモリからTopicへのpublishするbasler_example/imgpub.jsでできる。次のようにパイプで実行する。
~~~
./grabber | ./imgpub.js
~~~
imgpub.jsは共有メモリアクセスにshm-typed-arrayいうのを使いますが、これがまたパッチしないと動かないので、rovi/shm-type-arrayに置きました。
このJSはC++とリンクしているので、動作環境下でBuildが必要です。
~~~
node-gyp configure
node-gyp build
~~~

KazukiHiraizumi, 2018-04-02:
最終的には、ycam1h.jsからcamnodeを起動していますが、ここをgrabberを起動するように改変するだけだと思います。

KazukiHiraizumi, 2018-04-02:
Baslerのサンプルでは複数カメラでもプロセスは１個で捌かないといけないようです。YCAM-IIIはそういう構造なので、そっちに寄せてしまいましょう。

k-okada, 2018-04-06:
商用のライブラリを使う例です．
https://github.com/anqixu/ueye_cam/blob/master/CMakeLists.txt#L33

take-w, 2018-04-07:
> このJSはC++とリンクしているので、動作環境下でBuildが必要です。

手元だと
node-gyp configure
時に、

module.js:549
    throw err;

Error: Cannot find module 'graceful-fs'

というエラーになりますね。

~/catkin_ws/src/rovi/shm-typed-array/
でnpm install graceful-fsしても、
sudo npm install graceful-fs --globalしても、
同じ。
（npm listではgraceful-fsは見えるが。。）

~
だとnpm install graceful-fs自体が成功したように見えて
npm listでgraceful-fsが出てこず。

とりあえずRoVI GitHubに入っているビルド後バイナリで共有メモリアクセス自体はできていますが。。
（grabberを直接ROS化して、grabber内から画像をTopicにPublishするようにして、共有メモリを使わないようにするかもしれません。）

take-w, 2018-04-07:
@KazukiHiraizumi @Kan-Harada 
とりあえずSentechカメラドライバが動くように作れたのでpushしておきました。
（コミットメッセージに書き忘れましたがカメラの切断検知は未実装です。
　Sentechサンプルに沿ってStApi::IstDevice::IsDeviceLost()を使えば実現できるはず。
　でも後回し。）

上記の
> このJSはC++とリンクしているので、動作環境下でBuildが必要です。
は未解決です。
（まぁバイナリで動くので。。）

今後の予定は↓です。
1. RoVIでのcamnodeをgrabber, imgpub.jsに置き換え、SentechカメラドライバでRoVIが動くようにする。
2. 位相シフト組み込み。（←ここまで動けばひとまずRoVI全体の動作としてはOK）
3. Sentechカメラドライバ改良。（カメラ切断検知や上記JSのBuild問題など）

take-w, 2018-04-07:
あぁもちろん、camnodeとは違って、10FPSなら左右カメラからそれぞれ約100msec周期でライブ画像が入って来る、という正しい動きになっています。

（
TriggerMode=Onでのソフトウェアトリガとハードウェアトリガは動作未確認。
ソフトウェアトリガはRoVI Runモードでは使わない。（検査モードの静止画撮影では使うが後日。）
ハードウェアトリガは位相シフト組み込みで確認することになる。
）

KazukiHiraizumi, 2018-04-07:
@take-w 
スタンドアロン用にpshift...とycam1s.js,runner.jsをpushしました。pullしてください。
カメラドライバ名はycam1s.jsの55行目を修正してください。
カメラのパラメータ変更は、pshift...parseサービスにcsetというのを追加しました。
~~~
.../parse 'cset {パラメタ:値}'
~~~
nodejsのBuildが出来ない件は、今度のDRの時に。
ROS化するとパラメータの扱いもやらないといけないのでねえ。。。そこはC++で書くとムダに難しいので、JSで書
きたいです。

take-w, 2018-04-08:
@KazukiHiraizumi 
https://github.com/YOODS/rovi/blob/develop/yaml/rovi_param.yaml
を更新しました。

複数カメラでもgrabberプロセスは１個なので、左右カメラのパラメータは共通で
live:
  camera:
    AcquisitionFrameRate: 10.0
のようにまとめています。
（「live」部分はtwodでもgeneralでもdefaultでも何でもよい。）

別の言い方をすると左と右で別のGain値を設定するなどは不可能になる。それが望ましいはず。
（もともとcamera_aravisが左右で2プロセスでparamが分かれるからユーザが左右で別のGain値をrosparam setできて嫌だなぁ、、という話があり。）

live:
  camera:
    Width:
    Height:
を追加するかもです。
（いまはRoVI内で1280, 1024固定。）

> パラメータはGeVの素のパラメータ名に変更する。

http://www.emva.org/wp-content/uploads/GenICam_Package_v3_0_2.zip
を展開した
GenICam_Package_v3_0_2\SFNC\GenICam_SFNC_2_3.pdf
が最新のSFNCらしく、これに従っています。

よって
ExposureTimeAbsはExposureTimeへ。

それと、sentech grabberの作り上、DataStreamを作ってacquistion Threadを起動してから、
Gainなどをstdinから受け付ける、、となっていて、
これまでのmtu 9000パラメータはSFNCとしてはGevSCPSPacketSizeなのですが、
これはDataStreamを作った後に書き込みはできないようなので、
yamlパラメータからはmtuは消しました。
（grabberの起動時にgrabber内で9000を設定しています。）

take-w, 2018-04-08:
> スタンドアロン用にpshift...とycam1s.js,runner.jsをpushしました。pullしてください。
> カメラドライバ名はycam1s.jsの55行目を修正してください。
> カメラのパラメータ変更は、pshift...parseサービスにcsetというのを追加しました。
pullして、pshift...からgrabberまでつながるようにしてpushしました。

csetでyaml Parameterもgrabberへ投入してカメラに反映しています。

KazukiHiraizumi, 2018-04-09:
~~~
cam_l:
  camera:
~~~
にはID以外なくなった！ということですね。
ならいっそ、liveの下にgrabberをそのまま書けば、他のgrabberに変わっても対応しやすいのか？
~~~
live:
  grabber: "sentech_grabber STC_BBE132GE(17AB755) STC_BBE132GE(17AB756)"
  camera:
~~~

KazukiHiraizumi, 2018-04-09:
@take-w 
sentech_grabberのmakeでリンクエラーになるのですが・・
~~~
STAPI_ROOT_PATH=/opt/sentech make
~~~
この結果がこうなります。
~~~
g++ grabber.cpp cparser.c -o grabber -Wall -Wextra --pedantic -Wno-unknown-pragmas -Wno-switch -Wno-unused-but-set-variable -Wno-pointer-arith -Wno-unused-parameter -std=c++11 -I/opt/sentech/include/GenICam -I/opt/sentech/include/StApi -L/opt/sentech/lib -L/opt/sentech/lib/GenICam -lGCBase -lGenApi -lpthread -lStApi_TL
/usr/bin/ld: -lstdc++ が見つかりません
collect2: error: ld returned 1 exit status
Makefile:23: ターゲット 'grabber' のレシピで失敗しました
~~~

take-w, 2018-04-09:
まず、Sentech SDKはインストールされていますか？

/opt/sentech/share/doc/quick-start-ja.txt
に沿ってンストールと設定とサンプルのビルド(Make経由かCMake経由）はできているでしょうか。

沿っている場合、
STAPI_ROOT_PATH=/opt/sentech make
と言わず
make
だけでいけますが。。

（/usr/bin/ld: -lstdc++ が見つかりません
ということはSentech SDK以前の問題かも？）

take-w, 2018-04-09:
ちょっといまgrabberにデバッグ入れたりしてごちゃごちゃしてるので、もう少し落ち着いたらgrabberバイナリを送ります。

あるいはPC環境限定、ということでバイナリをgit pushした方がいいですかね？
（それでも結局はユーザもSentech SDKをインストールしないと実行はできないはずですが。。
　/opt/sentech/lib/lib*.soを使っているので。）

KazukiHiraizumi, 2018-04-09:
とりあえずそいつは、リンクを張ったらできました。
~~~
cd /usr/lib/x86_64-linux-gnu/
sudo ln -s libstdc++.so.6 libstdc++.so
~~~

take-w, 2018-04-09:
>  grabber: "sentech_grabber STC_BBE132GE(17AB755) STC_BBE132GE(17AB756)"
> argに()あってよかったのか

runner.jsからshで直接実行しているので、 ()があるとだめですね。
grabber: "sentech_grabber 'STC_BBE132GE(17AB755)' 'STC_BBE132GE(17AB756)'"
のように'でくくることになります。

take-w, 2018-04-09:
> 単体テストはこうですか？Exception出ますが・・・
> ./grabber SENTECH-17C8076 SENTECH-17C8077

引数のカメラはDisplayNameです。
（/opt/sentech/bin/StViewerで右下に見える。）

StViewerで確認するのが確実ですが、
./grabber 'STC_BBE132GE(17C8076)' 'STC_BBE132GE(17C8077)'
でいけると思います。

# SENTECH-17C8076などはarv-tool-0.4での表示ですよね？
# それはaravisがVendorとSerialNumberを - でくっつけて作っているようです。
# (VendorとSerialNumberもStViewerの右下に見える。)
# この辺は、カメラを一意に指定できればなんでもいいわけですが。。

KazukiHiraizumi, 2018-04-09:
了解。こちらのカメラでも単体テストOKです。

KazukiHiraizumi, 2018-04-09:
動きました。でもimage_viewで見ようとしたら
~~~
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Reinstalling the application may fix this problem.
~~~
エラーが出だしました。Fix方法わかりますか？

take-w, 2018-04-09:
どこにそのエラーが出ているでしょうか？
こちらでは見たことがなく。
（grabberが出しているのか…？）

最新のgitソースからgrabberをビルドして
roslaunch rovi run.launch
すると
launch窓にWARNレベルで
cam_l/image published
とか出るようにしているのですが、それは正しく出ていますか？

KazukiHiraizumi, 2018-04-09:
エラーはimage_viewからのよう
rvizで見ようとしても同じエラーが出るので、どうもQtのライブラリが壊れた？ような・・・・

closed 2019-02-16 by KazukiHiraizumi