YOODS/rovi #19: grabberでも画像が重なって来る

KazukiHiraizumi, 2018-04-09:
sentechのSDKを使ったgrabberでも、カメラ２台では重複画像が来る。
この現象はUDPの衝突が起きていることが原因で、aravisのせいではなかったのかも・・・

KazukiHiraizumi, 2018-04-09:
３．が正解ですね。PIC書き換えます。
これなら、パケット送出ディレイ(GevSCPD？？）とやらいうやつを正しく調整すれば、ライブも位相シフトも
同じになる。

Kan-Harada, 2018-04-09:
これって、「カメラに送るトリガを1フレーム分ずらす」、ということでしょうか。位相シフトの場合、プロジェクタとの同期が必要なで、それは無理と思いますが・・・勘違いでしょうか?

KazukiHiraizumi, 2018-04-09:
@Kan-Harada 
Issue#11にも上げてますが、
https://www.baslerweb.com/fp-1481127658/media/downloads/documents/application_notes/AW00064902000_Controlling_Packet_Timing_With_Delays.pdf
あたりの話ではないかと推測です。
対策として左カメラの１パケット(＝MTU)分の送出時間だけ右カメラの送出を遅らせる。
ハードトリガでもソフトトリガでも、この微妙なタイミング調整は無理ですが、外部トリガなら可能かな・・・
もし原因がそこなら１週間ロスですね。痛い・・・

KazukiHiraizumi, 2018-04-09:
Baslerのドキュメントでは,ドライバーにPacket Timeout(Inter-packet-delay)とやら設定があって、これでPacket間のintervalを監視しているらしい。

Kan-Harada, 2018-04-10:
先ほど中尾さんに確認したところ、windowsでリリースしているソフトウェアでは特に特殊なコーディング無しで同時トリガで動いているようです。

カメラ側にはディレイ設定をいれているらしらく、それが

> aslerのドキュメントでは,ドライバーにPacket Timeout(Inter-packet-delay)とやら設定があって、これでPacket間のintervalを監視しているらしい。

に相当するものでしょうか。それであれば、単にカメラ側に設定するだけで解決すると思われます。

Kan-Harada, 2018-04-10:
[dlp.pdf](https://github.com/YOODS/rovi/files/1892375/dlp.pdf)

DLPの動きを説明している資料ですが、実際問題としてDLPの発光とカメラトリガがどの程度同期しているかという確認作業が必要!! という話は別にして、下位1,2ビットの発光を捨ててしまうことにはなりそうです。

実際には発光パターンは5,6ビットあれば十分な気もしますが・・・

KazukiHiraizumi, 2018-04-10:
Basleのドキュメントだと、同一ネットに２台以上のカメラがあるときは、送出遅延を設定してパケット衝突が起きないようにしましょう、ということです。
カメラのトリガーは同時、i.e. CMOSの露光も同時ということが前提で書いてあります。
なのですが、TriggerDelayという設定はあるのですが、そのパケットディレイ的なものが見当たらない・・・

中尾さん：
Win版では、このディレイを何と言うパラメータを設定していますか？

closed 2019-02-16 by KazukiHiraizumi