YOODS/rovi #22: rvizについての話の場

take-w, 2018-04-16:
- 点群ファイルを見るならmeshlabでいいじゃないか、

- ROSで点群を見るだけなら
http://pointclouds.org/documentation/tutorials/cloud_viewer.php
でいいじゃないか、

という説もありますが、

rvizで点群を見るといろいろ便利なはず、、ということでrvizを使っているわけで、

使い方TIPSや「こんなことできないか？」みたいな情報を共有するための場を作りました。

私から1つ情報募集：
rvizで点群を自動フォーカスする方法はないか？
（RoVIの位相シフト計算ライブラリが点群座標をx=500, y=500, z=6000などの大きな値で返してくるので、マウス操作では点群を見つけられない。
　ズームアウトすると点群が小さくなって見えなくなる。
　なので一時的にgenpc_node.cppでのcoord設定で /100 とかしたり、
　あるいはそこを /10にしてrvizでSize (m)を大きな値にしたり、
　とかで無理やり見つける必要がある。）

take-w, 2018-04-16:
ちなみにRoVIでは位相シフト計算ライブラリが点群の色情報も返してくれるので、色情報も含めて/rovi/pcに点群をpublishしている。

この色情報もきちんとrvizで見るためには、rvizではColor TransformerをRGBF32にするのが正しい。
（ただ、どうせ元画像がMONO8で灰色だし、
　これまで通りrvizでIntensityでも見えるので気にしなくてもいい。）

Kan-Harada, 2018-04-16:
x,yは妥当な線として、z=6000は大きいですね。

とは言え、レクティファイしてない画像で試しているので仕方ない状況です。きちんとレクティファイした画像で位相シフトを呼べばカメラ座標系出力なので実距離に応じたz=350mm近辺に出てくるはずです。

KazukiHiraizumi, 2018-04-16:
Rvizの使い方は@k-okadaへ投げるのもよし
つまらないことでもどんどん質問してください、いうことでしたので・・・

k-okada, 2018-04-16:
１．pcl_viwer (http://pointclouds.org/documentation/overview/visualization.php
でpcd_viewer となっているもの）がapt-get install pcl-tools
すると使えるので，静的なファイル（PCDフォーマット）などになっていれば，これを使うことができるとおもいます．
２．ROS的にはPointCloud2でpublishしてrvizで見るのが良いです．
３．点群の単位系は[meter]になっていることが期待されているかとおもいます．
http://www.ros.org/reps/rep-0103.html#base-units
4. 色情報はkinectなどは以下のようになっていることが多い気がします．その場合RGBF32ではなくRGBを選択することになると思うのですが，，，
5.
とはいえ，将来的には超ラージスケールのpointcloudで全体が見たいなども有るかもしれないとはおもいます．pcl_viewerは全体がカバーできるように視点を変更する機能がありますので，同じものをrvizにいれてもいいかもとおもいました．bagファイルをどこかにおいていただければ，試してみます．
https://github.com/PointCloudLibrary/pcl/blob/master/visualization/src/interactor_style.cpp#L777

```
height: 480
width: 640
fields:
  -
    name: "x"
    offset: 0
    datatype: 7
    count: 1
  -
    name: "y"
    offset: 4
    datatype: 7
    count: 1
  -
    name: "z"
    offset: 8
    datatype: 7
    count: 1
  -
    name: "rgb"
    offset: 16
    datatype: 7
    count: 1
is_bigendian: False
point_step: 32
row_step: 20480
data: [0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 128, 63, 0, 0,
192, 127, 233, 127, 0, 0, 112, 89, 180, 202, 233, 127, 0, 0, 0, 0, 192,
127, 0, 0, 192, 127, 0, 0, 192, 127, 0, 0, 128, 63, 0, 0, 192, 127, 233,
127, 0, 0, 112, 89, 180, 202, 233, 127, 0, 0, 0, 0, 192, 127, 0, 0, 192,
127, 0, 0, 192, 127, 0, 0, 128, 63, 0, 0, 192, 127, 233, 127, 0, 0, 112,
89, 180, 202, 233, 127, 0, 0, 0, 0, 192, 127, 0, 0
```

closed 2019-03-23 by KazukiHiraizumi