YOODS/rovi #25: OpenCVバージョン問題

KazukiHiraizumi, 2018-05-02:
@Kan-Harada
ROSのインストール先(/opt/ros/kinetic/lib)にOpenCVのContribも含めインストールされているので、これでリンク出来ないとは、ちょっと考えられないです。

Kan-Harada, 2018-05-02:
了解しました。OpenCVのFileStorageクラスあたりにひっかかっているぽいので、この周りをもう少し調べてみます。それでもaptでインストールしたOpenCVではだめな場合はFileStrageを使わず、自力でFileI/Oを書く等で回避するようにします。

Kan-Harada, 2018-05-02:
よくわからないトラブルでしたが、位相シフトライブラリは/usr/libにインストールされているOpenCVをリンクするようにMakefileを書いていました。

ROSで使っているOpenCVが/usr/libのOpenCVではなく、「/opt/rosにインストールされているOpenCVである」ということで、include, libの場所指示を変えてみたところ、問題なくコンパイル・リンクは通ります。

/usr/libにaptでインストールされているものはOpenCV2.4ベースです。/opt/rosにインストールされているものはOpenCV3.3です。これがリンクエラーの原因ですね。

※PCの場合はこれで解決ですが、JETSONの場合
JETSONにJetpackでインストールされるOpenCVは/usr/libに入り、GPU対応になっています。一方、自分でソースからmakeするOpenCV3.3もGPU対応にしてあるので、いずれにしてもJETSON上で使う場合はこの後者の対応が必要になります。

KazukiHiraizumi, 2018-05-06:
rovi/lib/x86_64以下のlibopencv*が200MB以上あって結構エグいですが、これは消して大丈夫ですね。

KazukiHiraizumi, 2019-02-16:
Kineticは3.3.1-dev（3.4）

closed 2019-02-16 by KazukiHiraizumi