YOODS/rovi #27: ycam3vgaのlaunchでたまに /rovi/"regw" not available と出て起動しない

take-w, 2018-05-21:
@KazukiHiraizumi 
camera_aravisのサービスが準備出来ていないというエラー？

↓こういう感じです。

take@take-NXC154:~/catkin_ws$ roslaunch rovi run-ycam3vga.launch
[WARN] [1526897554.236] (ros): No camera params for L/R
[ERROR] [1526897559.260] (ros): /rovi/"regw" not available
[ERROR] [1526897559.261] (ros): Failure in openCamera
[rovi/pshift_genpc-6] process has died [pid 4381, exit code 101, cmd /home/take/catkin_ws/src/rovi/script/pshift_genpc.js ycam3 __name:=pshift_genpc __log:=/home/take/.ros/log/7a4e62ea-5cdf-11e8-b64e-a434d98edfe6/rovi-pshift_genpc-6.log].
log file: /home/take/.ros/log/7a4e62ea-5cdf-11e8-b64e-a434d98edfe6/rovi-pshift_genpc-6*.log
^C
take@take-NXC154:~/catkin_ws$

KazukiHiraizumi, 2018-05-22:
rosserviceでみても/rovi/camera/regwがいてないですか？

KazukiHiraizumi, 2018-05-23:
CreateStreamあたりで無限ループてるっぽい

take-w, 2018-06-27:
> CreateStreamあたりで無限ループてる
そのようです。

take-w, 2018-06-27:
> YCAM3D-IIIの電源をOFF->ONすると直るようです。

直らないことも多いようです。

take-w, 2018-06-27:
（現在開発用にPCでやってますが）
常にPC再起動後初回だけOKで、以降はPC再起動するまで常に
/rovi/"regw" not available
となる、、
という状況でした。

で、よく調べてみると、Issue #31の対応として入れた
> 現在はlaunch/ycam3vga.launchの先頭でsetgev.shを実行して
> 　arv-tool-0.4 control GevSCPSPacketSize=9000

の影響でそうなっています。

RoVIを一回実行して停止した後、
次回RoVIの先頭での
> arv-tool-0.4 control GevSCPSPacketSize=9000

を実行すると、その次のcamnodeが「CreateStreamあたりで無限ループ」です。

この
> arv-tool-0.4 control GevSCPSPacketSize=9000

は
arv-tool-0.4 control [○○]
arv-tool-0.4 features
arv-tool-0.4 genicam
などでも無限ループになります。

arv-tool-0.4 --help
の場合は無限ループにならずOK

take-w, 2018-06-27:
arv-tool-0.4 control GevSCPSPacketSize=9000
はIssue #31の
> （★最終的な解決としてはYCAM3D-III側の出荷時デフォルト値を9000にする必要がある。）

によって不要にはなるのですが、
パラメータ（DKRPQなど）をYCAMのgenicam XMLから取得する、
という話になっていて、そうすると
> arv-tool-0.4 genicam
でやはりおかしくなります。。

take-w, 2018-06-27:
↑けど、

arv-tool-0.4 genicamはcamnode起動前に打つと
（arv-tool-0.4 controlと同様に）今回の問題が起こるが、

camnodeがつながった後に打てば問題は起こらずにgenicam XML取得＆regwなどOK、

でした。

take-w, 2018-06-27:
不思議なことに、
Jetsonでは
arv-tool-0.4 control GevSCPSPacketSize=9000
をcamnode起動前に打っても問題起こらないですね。
（Jetson再起動など不要。）

PCでは再起動しないといけない、となる。
（なのでPCの場合は
arv-tool-0.4 control GevSCPSPacketSize=9000
を一時的にコメントアウトして実装は進められる。）

⇒しかし
>（★最終的な解決としてはYCAM3D-III側の出荷時デフォルト値を9000にする必要がある。）

がうれしい。

take-w, 2018-06-27:
PCでのこの再起動しないと直らない問題は別途Issueにします。

（本来のこの#27の「たまに」/rovi/"regw" not available）はこのままこの#27で。

closed 2019-03-23 by KazukiHiraizumi