YOODS/rovi #28: カメラIP設定(YCAM3D-III)

Kan-Harada, 2018-06-08:
https://github.com/areaDetector/aravisGigE/blob/master/etc/pulnix/forceip.py

これでUDPパケットを作ってIP設定できることは確認しました。事前にarv-toolを使ってカメラのMACアドレスを取得しておく必要があります。

但し、設定しても電源OFFで消えるので・・・どうやって保存できるのか。

KazukiHiraizumi, 2018-06-08:
https://github.com/areaDetector/aravisGigE/blob/master/COPYING
に書いてあります。商用でもFreeのようです。
macアドレスを固定にするか、筐体に表示するかですね。
製品表示は販売が前提ならどの道いりますが、未だそんな感じでもないなら、固定の方でどうでしょう。

KazukiHiraizumi, 2018-06-08:
Copyrightの条文をRoVIのどっかで引用や表示できるようにしないといけないのか、そのあたりは？？
ソフトコンソに聞くのがいいですね。

Kan-Harada, 2018-06-08:
販売する前提で部品在庫も抱えつつあるので、「IPアドレス設定機能」は必要と思って作業しています。
起動の都度、RoVIから設定する、というのもアリですが。。。。

Kan-Harada, 2018-06-24:
OKIのファームに手を入れてGVCPによるフラッシュメモリ内の書き換えができるようになったのでクローズします。

GVCPにより
・カメラのDHCP/固定IP変更が可能になった
・固定IPの設定が可能になった

同様の手順でカメラパラメータを書き込んで、RoVI起動時にレクティファイテーブル等を生成するようにしましよう。(現状YAMLファイル渡し)

closed 2018-06-24 by Kan-Harada