YOODS/rovi #35: camera_aravisのビルドが失敗する環境がある？

take-w, 2018-07-14:
Kan-Harada環境では（新しいROSワークスペースとして）

mkdir -p ~/catkin_ws2/src
cd ~/catkin_ws2/src
git clone https://github.com/YOODS/camera_aravis
cd ~/catkin_ws2
catkin_make

のようにして、（roviやEigenなしで）camera_aravisだけをビルドしようとすると、
~/catkin_ws2/devel/include/camera_aravis/GevRegs*.h
が見つからなくてビルドエラーになるとのこと。
（本来はGevRegs*.hはこのcatkin_make処理の中で作られ、正しくビルドが通るのだが。。）

roviやEigenを入れた後にまとめてcatkin_makeすると、camera_aravisも含めてすべて正しくビルド成功するとのこと。

他の環境でも同様なら、調べて解決する必要がある。
（Kan-Harada環境だけなら放置。）

Kan-Harada, 2018-07-14:
多分、同じことをやれば同じになると思うので、スッピンのUbuntu16.04(Desktop)でインストールに使ったコマンドをhistoryから記載しておきます。参考にしたのは下記サイトです。

http://wiki.ros.org/kinetic/Installation/Ubuntu

~~~
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
sudo apt-get update
sudo apt-get install ros-kinetic-desktop-full
sudo rosdep init
rosdep update
echo "source /opt/ros/kinetic/setup.bash" >> ~/.bashrc
source ~/.bashrc
sudo apt-get install python-rosinstall python-rosinstall-generator python-wstool build-essential
~~~

Kan-Harada, 2018-07-14:
Tajimaさん、ありがとうございます。CMakeLists.txtの最後にadd_dependenciesの行を追記したところ、うまくいきました。不勉強なため、まだ内容を理解していませんので、これからソースツリーをみてみます。

closed 2019-03-23 by KazukiHiraizumi