YOODS/rovi #36: Publish->Subscribeのスループット

KazukiHiraizumi, 2018-07-16:
1.3MB画像データをloopbackにてpub-subするスループットを計測。

|time to publish|time to subscribe|diff|
|:----|:----|:----|
|1531621775.659798|1531621775.670|0.010|
|1531621916.166472|1531621916.177|0.011|
|1531621935.132970|1531621935.142|0.009|

約１００MB/sくらいのスループット
メモリ転送の1/10のオーダ

KazukiHiraizumi, 2018-07-16:
Pub-SubよりService callを多用しているので、この部分はどうせ外からは見えないことから、Imageメッセージの仕様を拡張してはどうか
1. 通常はImage構造体のdata要素がビットマップデータであるが、これを共有メモリキー(4byte)にする
2. Server側はメッセージを受け取って共有メモリキーでメモリをアサインする

k-okada, 2018-07-17:
共有メモリができるということは，同じマシンで動かしているんだとおもいますが，その場合はnodeletを使うのが良いとおもいます．
これで，シリアライズもせずに直にメモリコピーされるはずです．
https://github.com/ros-drivers/freenect_stack/blob/master/freenect_camera/src/nodes/freenect_node.cpp
https://github.com/ros-perception/opencv_apps/blob/indigo/src/node/standalone_nodelet_exec.cpp.in

KazukiHiraizumi, 2018-07-17:
Nodelet化は先週の実装委員会でも但馬さんからご指摘いただきました。
よくよく考えたら、C言語Onlyではなかったので、共有メモリー渡しを検討しているとこです。
未だ実装はしてないです。

closed 2019-02-16 by KazukiHiraizumi