YOODS/rovi #37: ROSのタイミング解析ツール

KazukiHiraizumi, 2018-07-16:
タイミングチャートを作りたいが、思ったようなものがない(VxWorksのツールのようなもの)。
ニーズはありそうだが・・・Topコマンドとかでできるのかな？

7675t, 2018-07-17:
「タイミングチャート」が，プログラムの最適化のためのプロファイル（プログラムのどの部分にどのくらい時間がかかっているか）を知りたいということであれば，まずLinuxのgprofツールを検討してみるのが良いと思います．

- https://myenigma.hatenablog.com/entry/20150109/1420813105

基本的にはコンパイル時に`-pg`オプションを付けて実行し，終了させてできるgmon.outというファイルをgprofコマンドで処理すると，各関数にかかってる時間が分かります．私は使ったことはないですが，gprof2dotという可視化ツールも便利そうです．

- https://github.com/jrfonseca/gprof2dot

KazukiHiraizumi, 2018-07-17:
コメントありがとうございます。
gprof使ってみます。
ただ今C以外の言語もミックスで実装しているので、まずはざっくり状態を知るのにメッセージの飛び交っている状況をレコードできるツールがあればと思ってます。
Publish(subscribe)のタイムスタンプはレコード出来そうですが、そこにサービスコールも同時にレコードできれば、大体の動作は把握できそうです。

7675t, 2018-07-17:
メッセージはrosbagでタイムスタンプ付きで記録できますが，サービスを記録することはbagはできないし，いまのところ，できるツールもないようなのですよね．

- https://answers.ros.org/question/219126/rosbag-on-srvs/
- https://github.com/ros/ros_comm/issues/250

もしやるとしたら，Serviceサーバのクラスを拡張して，呼ばれたらメッセージをpublishする仕組みを追加する，などがあるかと思います．

k-okada, 2018-07-17:
>    - ros/ros_comm#250 <https://github.com/ros/ros_comm/issues/250>

その場合はActionLibを使うのが良いとおもいます．ServiceCallをTopicベースに実装して（かつプリエンプトできるようにした）ものがActionLibなので，ServiceCallで実装しているものをActionLibに変更して困ることは無いとおもいます．

ただ，基本はロボットのデータ処理はデータフローでできるはず，というのがROSでもOpenRTMでも基本になっているので，ServiceCallで実装している部分が，本当にTopicにならないか再確認するとよいおもいます．同期をとりたい，などの理由があるんだとおもいますが，後段でうまくやる方法，というものがいくつか提供されているので，基本はそれとの組み合わせを前提に考えて大丈夫なはずだと思っています．
http://wiki.ros.org/message_filters#ApproximateTime_Policy

KazukiHiraizumi, 2018-07-17:
ActonLibいうのはノーマークでした。勉強します。
そうですね。私共の設計はROSの基本形からちょっと外れていますね。
全体の工程制御をする１本のプログラムと、複数の個別処理(画像処理など)をするモジュール、という構造になってます。
ロボットというよりPLCとかに近い構造です。

KazukiHiraizumi, 2018-07-24:
若干(2〜10ｍｓ)のディレイはありますが、まずはrqt_plotでtopicをレコードするという手段でやってみましょう。
このためにtools_nodeにlogger0~3というサービス(std_srvs::SetEnable)を追加しました。任意のタイミングでこれをコールすると即log0~3トピックにpublishします。true=1,false=0です。
Nodejsからコールするときは必ずawait付きでCALLしてください。
~~~
await logger.call(req);
~~~
rqt_plotでは、FigureOptions->Curvesタグ->DrawStyleをStep（Post)に選択します。これでロジアナ風にTrue=High,False=Lowで表示されます。
[image]

KazukiHiraizumi, 2018-07-24:
ycam_ctrlでlogを３チャンネル使って以下を記録
1. 左カメラ系 imageイベント↑ remap.call↓ remap.return↑ イベントハンドラ終了↓ 
2. 右カメラ系で同じく
3. genpcサービス開始↑ 13枚Promise↓ Promise終了↑ genpc.call↓ genpc.return↑ サービス終了↓

KazukiHiraizumi, 2018-07-24:
[image]

k-okada, 2018-07-25:
そうですね．一通りできたところで，１回どこかで全体像を紹介してもらえればとおもいます．

利用者目線からすると，標準的な作り方にしておく他のパッケージとの類推が効くので，問題あったら直そうかな，というハードルが低くなります．
逆に特殊な作り方にすると，ユーザからはほぼブラックボックスと同じ形で接することになるかとおもいます．
なので，どのぐらいのところは，可能であれば，ユーザからのフィードバックももらいながら進めたいな，という風に考えて，
どこからは，ここは自分たちでやるから，大丈夫，と考えるか，というのがあるかとおもいます．

もう一個はリリースする際の問題で
https://github.com/YOODS/camera_aravis/
みたいに更新したものがあるとすると，オリジナルにpull requestしてリリースしてもらうところまでやっておくか，
自分たちのソースの中に取り組んじゃうか，あるいは，今回はtx2みたいなPCに組み込んじゃえば，外部のパッケージとは
別に考えてすすめるということも可能かもしれないとか
(repositoryを自社のものにして一般のros repositoryとは切り離す．ロボットーメーカはだいたいこれをしているはず．
ただ，組み込みPCに入って作業したくなると，一般の ros repositoryに変換してトラブルになる）

みたいなことが，よくある話題かな，とおもいます．

take-w, 2018-07-26:
[BSDライセンスについて2点相談質問.txt](https://github.com/YOODS/rovi/files/2232278/BSD.2.txt)

横からすみません、この#37の本筋とは違いますが、
「リリースする際の問題」「自分たちのソースの中に取り組んじゃうか」
というお話が出たのでここで相談させてください。
上記txtファイルの2点について、情報ございませんでしょうか。

take-w, 2018-07-26:
[stereo_image_procの問題点.txt](https://github.com/YOODS/rovi/files/2232286/stereo_image_proc.txt)

関連した内容を貼ります。
（特に重要な内容ではないので、無視していただいても構いませんが。。）

stereo_image_procで非平行カメラのdisparity計算にバグがあるが、
そもそもの意図がわからない部分があるし、
英語で突撃するのも大変だしなぁ、、ということで
RoVIではstereo_image_procを参考にdisarityなど実装している、ということですが。
（stereo_image_procを改良する…非平行カメラに正しく対応させる…ための情報になるtxtだとは思います。）

k-okada, 2018-07-26:
1. ライセンスについて．
```
【相談質問１】
ROSパッケージとしてのRoVIを同ライセンスにするには、
以下の実行が必要十分と理解していますが正しいでしょうか？

1. package.xmlのlicenseタグにBSDを指定する。

2. 各ソースコード（.h, .cpp, .js, .pyなど）に
   著作権表示やBSDライセンス条項を書く。
   （ただし.msg, .srvには書かない。）

   具体的には例えば.cppには以下のように書く：
   （Copyrightと、3条目の組織名をRoVI用にしている）
```
ROS的にはpackage.xml のlicense タグにライセンス名をつけます．licenseタグは複数つけることができます．
https://github.com/k-okada/warehouse_ros_sqlite/blob/jade-devel/package.xml

github的には最近は，リポジトリ作成時にトップディレクトリにLICENSEファイルを自動で置いてくれるようになった気がします．
https://github.com/tork-a/jog_control/blob/master/LICENSE

各ファイルのトップにライセンス条項を書くのは，昔ROSのところで，ファイルだけをどこか別のプロジェクトにコピーして使うときに，
こうなっている方が親切だよね，という議論があり，昔からROSを使っている人の慣習になった，という話を聞いたことが有ります．

```
【相談質問２】
現状、RoVIでは
https://github.com/YOODS/rovi/blob/develop/src/sgbm_node.cpp
の中で、
stereo_image_procパッケージ
http://wiki.ros.org/stereo_image_proc
https://github.com/ros-perception/image_pipeline/tree/indigo/stereo_image_proc
の一部ソースをコピペして変更して使用しています。
（ちなみに、
　なぜstereo_image_procパッケージをそのままnode(let)として使わないのか
　についての詳細は「stereo_image_procの問題点.txt」に記述します。）

具体的には、
https://github.com/YOODS/rovi/blob/develop/src/sgbm_node.cpp
の
	49行目からのisValidPoint()が
	https://github.com/ros-perception/image_pipeline/blob/indigo/stereo_image_proc/src/libstereo_image_proc/processor.cpp
	の同名関数のそのままコピペ。

	233行目
	model_.fromCameraInfo(linfo, rinfo);
	から
	300行目
	pub_disp.publish(disp_msg);
	までが
	https://github.com/ros-perception/image_pipeline/blob/indigo/stereo_image_proc/src/nodelets/disparity.cpp
	のDisparityNodelet::imageCb()の中味をコピペして
	その中の if (cx_l != cx_r) {} を消したもの。

となっています。

stereo_image_procはROSコアパッケージの一部で（修正）BSDライセンスです。

このような場合、
stereo_image_procのソースの一部を使っていることを
RoVI内に明記する必要があると思うのですが、
具体的にはどこにどのように記述すべきか、
情報をいただけないでしょうか？
```
うーん．なるほど．直感的にはBSDの宣伝条項がなくなっているから，これは何も書かなくても良い，ということになるでしょうか？ 自信ないので，@7675t さんどうでしょうか．

stereo_image_procの問題点.txt
```
実際のソースでは
		d = x_l - x_r - (cx_l - cx_r)
		ではなく
		d = x_l - x_r - 2 * (cx_l - cx_r)
	となってしまっている。
```

https://github.com/ros-perception/image_pipeline/blame/8d01c930e66bdeb58c4b138539a64ddea0b17e10/stereo_image_proc/src/nodelets/disparity.cpp#L202-L209

流石に８年前のコードでわかる人は誰もいない状況ですが，確かに変なのかな．．．
  // Adjust for any x-offset between the principal points: d' = d - (cx_l - cx_r)
`d = x_l - x_r `は両方cx=0 の時の話ではないでしょうか．

いずれにせよ，
```
・stereo_image_procをForkして自分たちで直す。（そしてPull Requestして本家へ反映してもらう。）
のどちらかが正統だと思うが、
```
の一択だとおもいます．
コメントは最初はそんなに真面目に書かなくても
```
this code calculate   
		d = x_l - x_r - (cx_l - cx_r)
not
		d = x_l - x_r - 2 * (cx_l - cx_r)
because of 
https://github.com/ros-perception/image_pipeline/commit/eba97a1a919b7f2d9a30991d46da8e2a9317341b#diff-008447bf1f554d6a9a1431cff3ff1bcbR70
and
https://github.com/ros-perception/image_pipeline/blame/8d01c930e66bdeb58c4b138539a64ddea0b17e10/stereo_image_proc/src/nodelets/disparity.cpp#L202-L209
```
ぐらいでいいとおもいます．

本当は簡単に問題が再現できる例を作れると良いんだけど，，，，
プログラムで画像をpublishしたらいいのかな．

7675t, 2018-07-27:
量が少ないので微妙なところですが，コピーしたのであれば厳密に言うと著作権表示は必要だと思います．同じBSDライセンスで再配布する場合は，部分的に取り込んだ旨の著作権表示を表示すればいいかなと思います．

たとえば

```
* Software License Agreement (BSD License)
*  Copyright (c) 2018, YOODS, Inc.
*  Portions are:
*  Copyright (c) 2008, Willow Garage, Inc.
```
みたいな感じでしょうか．"portions are"で特に部分の指定なくどこか一部と言っていて，大半がオリジナルなので，付随的な感じで書いても良いかなと．「この関数は」みたいな切り分けができればそれでも良いかもしれませんが．

参考（実例が挙がってるだけなので，まあ慣習的には間違ってないとおもいます）
- http://d.hatena.ne.jp/mozxxx/20110529/p4

k-okada, 2018-07-27:
ナルホド。勉強になりました。ありがとうございます。

take-w, 2018-07-27:
お二方ともありがとうございます。
非常に参考になりました。😊

【相談質問１】
についてはpackage.xmlのlicenseタグにBSD、トップディレクトリにLICENSEファイルでBSD内容とします。

【相談質問２】
も今後は一部BSDコピペを元にした場合はこのようにします。
（下記Forkによって、今回はコピペはなくなります。）

> ・stereo_image_procをForkして自分たちで直す。（そしてPull Requestして本家へ反映してもらう。）

の方針で進めます。

take-w, 2018-07-30:
@k-okada @7675t 
すみません、また相談させてください。。

https://github.com/ros-perception/image_pipeline/pull/349
でPull Requestしてみましたが、
All checks have failed
となって、
https://travis-ci.org/ros-perception/image_pipeline/builds/409833703?utm_source=github_status&utm_medium=notification
で全滅しているが理由は4つとも
The command "sudo apt-get install -y python-catkin-pkg python-catkin-tools python-rosdep python-wstool ros-$ROS_DISTRO-catkin" failed and exited with 100 during .
となっていて、私のソースが悪いというわけではなさそうでして。。
たまたまapt-get先のサーバが何かおかしかったということ・・？
（その前の他の人のPull Request
　https://travis-ci.org/ros-perception/image_pipeline/jobs/407422369
　では同コマンドが通っている。）
（もちろん手元のローカルPCではビルド通っています。）

このような場合、放っておいてもros-perception/image_pipelineの管理者がPull Requestを良きに処理してくださるものでしょうか？
それとも例えば再度Pull Requestしてみる、などの手段が必要でしょうか？

7675t, 2018-07-31:
@k-okada の方が詳しいので補足して欲しいです．

エラーについてはたまに起きます．理由は多分推察通りだと思います．
PRをリビルドする必要がありますが，レポの書き込み権限を持ってないと，travisの画面のrebuildボタンが現れないようです．

方法としては，PR元のレポに新たに変更をプッシュする，というのもありですが，無意味な変更とかはしたくないと思うのでイマイチですね．

- https://stackoverflow.com/questions/17606874/trigger-a-travis-ci-rebuild-without-pushing-a-commit

によると，一回Pull RequestをCloseし，もう一回Openすると，ビルドされるそうです．こちらのほうが良さそうに思います．

PRについてですが，メンテナの立場に立ってみると，なかなかMergeボタンを押しにくい内容だなあと思いました．変更自体はとても小さいですが，他のパッケージや利用者への影響が大きそうなので，大丈夫なんだろうか，と確信が持てないかもしれません．よっぽど画像処理になれた人なら，ああ間違えてるね，で押せるのかもしれませんが．

なので，すぐにはマージされないのではないかと思います．@k-okada の他のPRもけっこう長い間放置状態ですね．ただ，自分のレポに他の修正を取り込むのが容易なので，forkして修正していくのが良いとは思います．もしかしたら「使用前」「使用後」で，実際に改善されたデータを示せれば，助けになるかもしれない，と思いました．

k-okada, 2018-08-01:
Travisに関しては基本放置でいいとおもいます．https://github.com/ros-planning/moveit/pull/973#issuecomment-404101710 , https://github.com/jsk-ros-pkg/jsk_pr2eus/pull/352#issuecomment-409222194 みたいに自分のコミットとは違うところで，エラーになっていますね，というコメントが一言あれば，メンテナの人もすぐに対応はできるとおもいますが，そもそも，このリポジトリも（超基本的なパッケージにかかわらず）メンテが十分でなくて困ったところです．
ドライバなど，簡単であまり変更なさそうなものはOrphanedPackagesとして自分たちでメンテするところまでボランティア刷るようにしましたが（http://wiki.ros.org/OrphanedPackage#Kinetic）, `image_pipeline` まで手を出すかどうか，，，悩ましいところです．

で，もともとのhttps://github.com/ros-perception/image_pipeline/pull/349 ですが，手元で再現できるように人工のステレオ画像をpublishするノードを書いていて，あれ，っとおもったんですが，`cx_r != cx_l` は輻輳カメラの場合，ということなんだけど，これはrectify のところで平衡の画像に直されているはず，というふうに思ったんですが違いますでしょうか？

% 本当に１５年ぶりにrectify というコトバを使ったぐらいで，この前もなんかの3Dデータがカメラからの距離，だったので，xyz軸のz軸にそった距離か，視線方向にそった距離か，という話で完全に間違えていたりしているので，信用しないでください．

k-okada, 2018-08-01:
あ、すいません。やっぱ勘違いでした
輻輳しているとカメラをimage_procに通すと、画像は平行化されるけどcxは大きく違いますよね。と言うことでね

closed 2019-03-23 by KazukiHiraizumi