YOODS/rovi #38: 位相シフト（/rovi/pshift_genpc Service）での点群渡しやpublishヶ所について

take-w, 2018-07-25:
@KazukiHiraizumi 

＜現状の問題点＞
位相シフト（/rovi/pshift_genpc Service）では、以下の処理の流れになっている。

1. ユーザがサービス呼び出し（rosservice call /rovi/pshift_genpcなど）。
2. 同サービスを提供するscript/ycam_ctrl.jsが、src/genpc_node.cppの内部サービス（/rovi/genpc）を呼び出し。
3. src/genpc_node.cppは点群を計算し、【srv/GenPC.srvのresに点群を乗せて】return。
4. script/ycam_ctrl.jsは3の点群を受け取って、/rovi/pc, /rovi/pc2 TopicにPublish。

（この設計の背景には、4でycam_ctrl.jsが点群を受け取って、物体認識処理などにそのままつなげたい、ということがある。）

ここで、3の【〇】で乗せて4でycam_ctrl.jsが点群を受け取るまでに、点群数が多いと長い時間がかかっている。
（130万画素だと2, 3秒など。）

なお、Publish自体は3で行っても4で行っても一瞬で終わる。

★提案★
3で【〇】で乗せずに、3でPublishしてしまえば処理時間がかからずよいのでは？
4で物体認識処理などのために点群が欲しければ、（どうせycam_ctrl.js側で/rovi/genpc callするタイミングは制御できているので）/rovi/genpc callの前に3の点群topicをsubscribeし、それを使って物体認識処理などをして、その後unsubscribeすればよいと思う。

（#36の共有メモリの話や#37の「基本はロボットのデータ処理はデータフローでできるはず」or「ロボットコントローラというよりPLCとかに近い構造」とも関連。）

どうでしょうか🤔

KazukiHiraizumi, 2018-07-25:
当面、genpc_nodeでPublishしてServiceのResponseには乗せない、という仕様でいいと思います。
解せないのは、genpc_nodeへのRequestでは、１．３MB×26のデータを送信しているのに、これは問題にならない？ということです。
#37にも書いてますが、解析ツールでタイミングチャートを起こさないと、正しい判断が出来ないと考えています。

KazukiHiraizumi, 2018-07-25:
データフロー方式ということだと、genpcへは更新の度、1枚ずつ送っておいて、genpc側で判断してて点群計算する、ということになりそうですが、それはそれで難しい。
全部トピックに出ていると、Rosbagでレコードして解析できるというメリットはありますが、現時点では解析の方は別手段を検討するということにしましょう。

k-okada, 2018-07-26:
以下は普通のデプスカメラを使う時の一般論，ですが．．．

ROSのデプスカメラはDepthImageを出して，そのあとポイントクラウドに展開して利用するものが多いです．
理由はそういう仕組みのセンサである，という以外に以下にあるようにデータ量が1/8というメリットがあります．
http://wiki.ros.org/openni_launch/Tutorials/BagRecordingPlayback

カメラがつながっているPC（ロボットの体内PC)と，画像処理をしたいPC(体外のGPU付きPC）があるような
環境ではPointCloudは送らずにDepthImageを送るのが一般的です．

ただ，データ型がIntになるので，精度が落ちるのかもしれません．

DepthImageの作り方
http://jsk-recognition.readthedocs.io/en/latest/jsk_
pcl_ros/nodes/depth_image_creator.html
https://answers.ros.org/question/192680/depth-image-from-point-cloud/

ImageからPointCloudを作る
http://wiki.ros.org/depth_image_proc#depth_image_proc.2BAC8-point_cloud_xyz

KazukiHiraizumi, 2018-07-26:
情報ありがとうございます。データ量が1/8になれば要求値をクリアできます。
ノード間通信のデータ量に配慮が要りましたね。反省点です

closed 2019-03-23 by KazukiHiraizumi