YOODS/rovi #44: rovi構成について

KazukiHiraizumi, 2019-03-22:
昨日のRRI柏の葉MTGでReposの整理をということでした。以下でどうでしょうか？
## rovi/コア
<table>
<tr><th>Directory<th>File<th>Note
<tr>i<td rowspan="3">~<td>README.md
<tr><td>GettingStarted.md
<tr><td>Install.sh<td><i>[New]<i>apt化できるまでGettingStarted手順をshに
<tr><td rowspan="4">launch<td>ycam3sxva.launch
<tr><td>ycam3vga.launch
<tr><td>ycam3vga.launch
<tr><td>ycam3s.launch<td>直接launchしない。ycam3loader.shからlaunchされる
<tr><td>ycam3loader.sh<td>mm&rarr;m変換もここに要る
<tr><td rowspan="3">srv<td>Dialog.srv<td>roviコアで未使用なもの削除、移動
<tr><td>ImageFilter.srv
<tr><td>GenPC.srv
<tr><td rowspan="13">src<td>CalibBoard.h
<tr><td>CircleCalibBoard.h
<tr><td>ParaConfigure.h
<tr><td>PointCloud.h
<tr><td>UnionFind.hpp
<tr><td>camera.h
<tr><td>camera_calibration.h
<tr><td>common.h
<tr><td>phase_shift.h
<tr><td>ps_main.h
<tr><td>genpc_node.cpp
<tr><td>grid_node.cpp
<tr><td>remap_node.cpp
<tr><td rowspan="18">script<td>CalibBoard.h
<tr><td>GevForceIP.py<td>これ使えるの？
<tr><td>gvdump.js
<tr><td>gvgetid.js
<tr><td>gvload.js
<tr><td>image_switcher.js
<tr><td>impub.py<td><i>[Todo]</i>汎用的に作ること
<tr><td>imsave.js<td><i>[Todo]</i>汎用的に作ること
<tr><td>notifier.js
<tr><td>p2qmatrix.py
<tr><td>param_watch.js
<tr><td>rosrun.js
<tr><td>sens_ctrl.js
<tr><td>tflib.js<td>Install.shでNODE_PATHへコピー
<tr><td>tflib.py<td>Install.shでPYTHONPATHへコピー
<tr><td>ycam1h.js<td><i>[Todo]</i>YCAM1が使えるように
<tr><td>ycam3s.js
<tr><td>ycamctl.js
</table>

k-okada, 2019-03-26:
はい，まずはコア（＋utils)だけのパッケージを作ると良いとおもいます．

作った後で，一度，以下を確認できるとよいとおもいます．

１）標準のサービス，仕組みで代替出来る機能は無いか？
（標準の仕組みを使っておくと，経験者にとっては類推でシステムを使えるので便利になります，また，関連ドキュメント，ツールも既存のものがあるので手間が省けます）

srv/ Dialog, GenPC, ImageFilter あたりはどのような理由になるでしょうか？

Dialogが状態監視だと，あまりここまで丁寧にやっているdriverはないですが，
https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/src/driver/driver.cc#L131
みたいな仕組みを使うとrqt_robot_monitor で観察できます．
ロボットに組み込んだときに以下のように動いていなければ黄色や赤でマークがでます．
http://wiki.ros.org/diagnostics/Tutorials/Using%20the%20Robot%20Monitor

画像が大きいなどの理由で画像をストリームしない場合は
http://wiki.ros.org/polled_camera
という仕組みがあります．

depth_node.cpp	はdisparity画像を距離画像(depth)にするものでしょうか？
これは判断が難しいですが，Kinectのようにドライバがdepthまで出す，とうするのも一つかなともおもいます．
https://github.com/ros-perception/image_pipeline/issues/71
メンテされていないのでマージはされないとおもいますが，いかに新しいノードとして追加しても良いのかもともいました．
http://wiki.ros.org/depth_image_proc#depth_image_proc.2BAC8-convert_metric

以下は細かすぎるポイントです．

> 直接launchしない。ycam3loader.shからlaunchされる

https://github.com/fetchrobotics/fetch_robots/tree/melodic-devel/fetch_bringup/launch/include
みたいに階層を作るのと，suffixをlaunchでなくしておくとroslaunchしたタブを打った時の候補に現れないので，直接呼び出される可能性が減ります．という方法がたまに使われます．

KazukiHiraizumi, 2019-03-26:
@k-okada 
いろいろご指摘ありがとうございます。サービスについては既存のパッケージで代替えがありそうな気がしていたので調べてみます
最後のlaunchですが、これは本来一つのlaunchにまとめたいのですが、パラメータの前処理が終わってから起動しないといけないnodeがあり、仕方なくShellに分けて順序のインタロックをとっています。
launchで
<node type="A">
<node type="B">
とあたっとき、Aがある処理まで達してから（あるいは終了してから）Bを起動することはできないでしょうか？

k-okada, 2019-03-27:
それは出来ないです．
https://answers.ros.org/question/233353/set-delay-between-starting-nodes-within-launch-file/
ココに書いてある以外に，ものすごく汚い方法としては
http://wiki.ros.org/roslaunch/XML/param
のcommandをつかえばどのようなプログラムも事項でき，このparamというtagは全てのノードの起ち上げの前に行われるので，２つの処理しかなければ順番を規定できます．

ただ，基本的には手順に依存するようなプログラムを書いてはいけない，ということかとおもいます．

ところで，依存するものはどういうものでしょうか？パット見でなかなか追いかけられていませんが，
https://github.com/YOODS/rovi/blob/ramiel/launch/ycam3loader.sh
IPADDRSからguid をとってきて，それをパラメータにしてノードをたちあげているでしょうか？
たとえば，guidが固定なら，最初の１回はrosrun xx xxx してguidを調べてください，あとは，それを引数にしてnode を起ち上げてください，みたいな運用もあるかとおもいました．

https://github.com/ssafarik/camera_aravis
などはもう少し簡単になっているかとおもうので，ちょっと不思議におもいました．

KazukiHiraizumi, 2019-03-27:
キャリブレーション情報とカメラのペアリングが保てるように、弊社のカメラはキャリブレーション情報をyaml形式でカメラのFlashRomに書き込んで出荷しています。
起動前に、これを読出(gvdump.js)して、rosparamとかで展開しておきたい、という目的です。
カメラを変更した場合は、一旦コマンドを起動して、キャリブレーション情報をyamlファイルに書き出す、という使い方でも、いいようには思います。

k-okada, 2019-03-28:
それは，以下のような仕組みを使ってcamerainfoをサービスで掛けるようにしておいて，
flasahromから読みだしてrosparamで展開し，サービスを呼ぶのが標準的な書き方になるかとおもいます．
https://github.com/ros-perception/image_common/blob/hydro-devel/camera_info_manager/src/camera_info_manager.cpp
https://github.com/ros-drivers/video_stream_opencv/blob/master/src/video_stream.cpp#L299

closed 2019-08-06 by KazukiHiraizumi