YOODS/rovi #48: about ROS -kenitic install

MatuokaNaru, 2019-07-07:
in tarminal,

matuoka@matuoka-VirtualBox:~$ sudo apt-get install ros-kinetic-desktop-full 
E: ロック /var/lib/dpkg/lock-frontend が取得できませんでした - open (11: リソースが一時的に利用できません)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
matuoka@matuoka-VirtualBox:~$

MatuokaNaru, 2019-07-07:
in ubuntu, i dont use japanese

MatuokaNaru, 2019-07-07:
i can`t use japanese

MatuokaNaru, 2019-07-07:
after time,

matuoka@matuoka-VirtualBox:~$  apt-get install apt-file
E: ロックファイル /var/lib/dpkg/lock-frontend をオープンできません - open (13: 許可がありません)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?
matuoka@matuoka-VirtualBox:~$

MatuokaNaru, 2019-07-07:
after, 

matuoka@matuoka-VirtualBox:~$ sudo apt-get install ros-kinetic-desktop-full
パッケージリストを読み込んでいます... 完了
依存関係ツリーを作成しています                
状態情報を読み取っています... 完了
E: パッケージ ros-kinetic-desktop-full が見つかりません
matuoka@matuoka-VirtualBox:~$

Kan-Harada, 2019-07-07:
どういう訳かaptがロックされているようです。ちょっと調べてみただけですが

https://qiita.com/koara-local/items/e63b7e4ff3cea803bdea

を参考にロックファイルを削除してみては?

MatuokaNaru, 2019-07-10:
./Install.sh: 74 行: cd: /home/matuokanaru/*/src/rovi: そのようなファイルやディレクトリはありません
fatal: destination path 'rovi_utils' already exists and is not an empty directory.
./Install.sh: 79 行: cd: /home/matuokanaru/*/src/rovi: そのようなファイルやディレクトリはありません
fatal: destination path 'rqt_param_manager' already exists and is not an empty directory.
./Install.sh: 84 行: cd: /home/matuokanaru/*/src/rovi: そのようなファイルやディレクトリはありません
Base path: /
The specified source space "/src" does not exist
./Install.sh: 89 行: cd: /home/matuokanaru/*/src/rovi: そのようなファイルやディレクトリはありません
cp: 'launch/rovirun.sh' を stat できません: そのようなファイルやディレクトリはありません
./Install.sh: 91 行: cd: /home/matuokanaru/*/src/rovi_utils: そのようなファイルやディレクトリはありません
cp: 'lib/python2.7/*.py' を stat できません: そのようなファイルやディレクトリはありません

MatuokaNaru, 2019-07-10:
一応日本語化、できました。
あと、ROSのインストールができたのですが、roviのインストールができません。
./install.shで実行した結果です

MatuokaNaru, 2019-07-10:
matuokanaru@matuokanaru-VirtualBox:~/rovi$ source /opt/ros/kinetic/setup.bash
matuokanaru@matuokanaru-VirtualBox:~/rovi$ source ~/catkin_ws/devel/setup.bash
bash: /home/matuokanaru/catkin_ws/devel/setup.bash: そのようなファイルやディレクトリはありません
matuokanaru@matuokanaru-VirtualBox:~/rovi$ export ROS_HOSTNAME=localhost
matuokanaru@matuokanaru-VirtualBox:~/rovi$ export ROS_MASTER_URI=http://localhost:11311
matuokanaru@matuokanaru-VirtualBox:~/rovi$ export NODE_PATH=/usr/lib/node_modules
matuokanaru@matuokanaru-VirtualBox:~/rovi$ export PYTHONPATH=/usr/local/lib/python2.7/dist-packages:$PYTHONPATH
matuokanaru@matuokanaru-VirtualBox:~/rovi$

MatuokaNaru, 2019-07-10:
あと、ROSアカウントの初期設定ですが2行目において
そのようなファイルやデレクトリはありません。
が出てしまいました。

KazukiHiraizumi, 2019-07-10:
@MatuokaNaru 
cd が失敗してそう。
...src/のrovi以外のパッケージ消して、Install.shを以下のように変更し試してもらえますか？
1. 最初に  rovidir=$PWD  としてカレントディレクトリ(.../roviのはず)を保存
2. cd ~/*/src/roviをcd $rovidirに置き換え
3. 最後のcd ~/*/src/rovi_utilsはcd $rovidir;cd ../rovi_utilsに変更

KazukiHiraizumi, 2019-07-10:
アカウントの初期設定の失敗の方は、一回catkin_makeを走らせないと出ると思う。
でもcatkin_makeを走らすにはInstall.shが成功しないと。。。になっているので

Kan-Harada, 2019-07-17:
日報を見ると、うまくいったようですね。

MatuokaNaru, 2019-07-17:
そうです。一応友達と協力して終わりました

closed 2019-08-06 by KazukiHiraizumi