YOODS/rovi #49: RoVI mm単位の起動について

Reo-Take, 2019-07-17:
mm単位でのroviの起動について
README.mdでは

SXGAモード(mm単位)
roslaunch rovi ycam3sxga_mm.launch
VGAモード(mm単位)
roslaunch rovi ycam3vga_mm.launch

と書いてありますが，実際のファイル名は

ycam3sxga_m.launch
ycam3vga_m.launch

となっていました。
とても細かいことですが、一応、書いておきます。

Kan-Harada, 2019-07-17:
これはドキュメントが間違っていますね。実際にlaunchファイルの中を見てみてください。

SXGAモード(mm単位)
roslaunch rovi ycam3sxga.launch
VGAモード(mm単位)
roslaunch rovi ycam3vga.launch

SXGAモード(m単位)
roslaunch rovi ycam3sxga_m.launch
VGAモード(m単位)
roslaunch rovi ycam3vga_m.launch

が正解です。README.mdを修正してpull requestしておいてください。

Reo-Take, 2019-07-17:
すみません。pull requestが上手くいきません。
個人的にはGitHub上のリポジトリへの反映が上手くいっていないと考えています

reona@DESKTOP-C8ASVD4:~/catkin_ws/src/rovi$ vim README.md
reona@DESKTOP-C8ASVD4:~/catkin_ws/src/rovi$ git add README.md
reona@DESKTOP-C8ASVD4:~/catkin_ws/src/rovi$ git commit -m "起動について修正しました"
[israfel 8e1f2f0] 起動について修正しました
 1 file changed, 3 insertions(+), 3 deletions(-)
reona@DESKTOP-C8ASVD4:~/catkin_ws/src/rovi$ git push israfel israfel
fatal: 'israfel' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
reona@DESKTOP-C8ASVD4:~/catkin_ws/src/rovi$

KazukiHiraizumi, 2019-07-17:
@Kan-Harada
単位系ですが、ここだけでなく色んなとこ(キャリブ板、ロボット通信)とかにも係わってくるので、環境変数とかで設定した方がいいのかもしれないです。
柏の方は基本mです。
案件絡みはmmですね。
早めにやっておいた方がいいな。きっと・・・

Kan-Harada, 2019-07-17:
@Reo-Take 
1) 自分に一旦Fork->israfelにブランチ切替え
2) 自分用のブランチを作成->ブランチ切替え
3) 手元で修正
4) commit->push
5) pull request
という手順だと思います。

※単位系の話はこれだけにとどまらないようなので、ひとまずpull reqは無しで良さそうです。

Reo-Take, 2019-07-17:
pull requestの件、かしこまりました

今回のpull requestが上手くいかなかったことに関しましては、README.mdに書いてある

RoVIのインストール
ROSのソースディレクトリ(~/catkin_ws/srcなど)に、israfelブランチをcheckoutします。
git clone -b israfel --depth 1 https://github.com/YOODS/rovi.git

というコマンドをそのまま実行していたため、ForkされたものではなくYOODSのリポジトリがそのままcloneされていたことが原因だったみたいです。お騒がせしてすみませんでした。
一応、問題が解決されたか確かめたかったためpull requestをかけてみました。

KazukiHiraizumi, 2019-07-17:
@Reo-Take
m単位のときのlaunchファイルも修正が要ります
~~~
roslaunch rovi ycam3vga_m.launch
~~~

Reo-Take, 2019-07-18:
@KazukiHiraizumi 

すみません。見落としていました。
pull requestができるかどうか確かめるためにpull requestを行ったのですが、修正したほうがよろしいですか。

KazukiHiraizumi, 2019-07-18:
@Reo-Take
わかりました。PullReqは受け取りました。
こんな感じで送ってください。
この件については
>単位系ですが、ここだけでなく色んなとこ(キャリブ板、ロボット通信)に係わってくるので、環境変数とかで設定した方がいいのかもしれないです。
ように改変要りそうなので、理解が深まってきた段階でまたおねがいします。

Reo-Take, 2019-07-19:
@KazukiHiraizumi 

かしこまりました。

closed 2020-10-21 by KazukiHiraizumi