YOODS/rovi #55: rovi_master_teach/demo.launch実行について

MatuokaNaru, 2019-07-28:
先日送ったメールにおいて、順番にやったのですが
3Dスキャン実行→シーンがrvizに表示される
9) 解析実行　で以下のエラーメッセージが出てきました

[ERROR] [1564313954.741953]: bad callback: <function cb_solve at 0x7f86b96b0aa0>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/matuokanaru/catkin_ws/src/rovi_utils/script/searcher.py", line 135, in cb_solve
    solveResult=solver.solve(Scene,Param)
  File "/home/matuokanaru/catkin_ws/devel/lib/python2.7/dist-packages/o3d_solver.py", line 78, in solve
    res,pcs=match_()
  File "/home/matuokanaru/catkin_ws/devel/lib/python2.7/dist-packages/o3d_solver.py", line 19, in match_
    source=copy.deepcopy(modPcArray[0])
IndexError: list index out of range

どのように対処すればいいでしょうか？

MatuokaNaru, 2019-07-28:
ちなみにRVIｚでは青色の立体しか出てきません

Kan-Harada, 2019-07-29:
rovi_utilsのブランチがmasterになっているような気がします。
git checkoutオプションで切り替えるか、一旦、src/rovi_utils/を消して再度git clone -b devel ....で再度cloneしてみてください。

MatuokaNaru, 2019-07-29:
なんとか行けました。ありがとうございます

closed 2019-08-06 by KazukiHiraizumi