YOODS/rovi #58: catkin_makeが失敗する件

YukiYamagata, 2019-08-14:
rkt_toolsをclone及び、roviやrovi_master_teach等をgit fetch git mergeして、
catkin_makeを行ったのですが、途中でエラーになりました。
下記がそのエラーメッセージです。
camera_aravisで問題が起きているようだったので、
./rovi/Install.shを再度実行しても、改善されませんでした。
どうすれば改善できるでしょうか？

[エラーメッセージ]
ModuleNotFoundError: No module named 'rospkg'
camera_aravis/CMakeFiles/camera_aravis_gencfg.dir/build.make:63: ターゲット '/home/yuuki/catkin_ws/devel/include/camera_aravis/CameraAravisConfigConfig.h' のレシピで失敗しました
make[2]: *** [/home/yuuki/catkin_ws/devel/include/camera_aravis/CameraAravisConfigConfig.h] エラー 1
CMakeFiles/Makefile2:3867: ターゲット 'camera_aravis/CMakeFiles/camera_aravis_gencfg.dir/all' のレシピで失敗しました
make[1]: *** [camera_aravis/CMakeFiles/camera_aravis_gencfg.dir/all] エラー 2
make[1]: *** 未完了のジョブを待っています....
[ 65%] Built target camera_aravis_generate_messages_cpp
[ 70%] Built target camera_aravis_generate_messages_py
[ 70%] Built target camera_aravis_generate_messages_nodejs
Makefile:138: ターゲット 'all' のレシピで失敗しました
make: *** [all] エラー 2
Invoking "make -j4 -l4" failed

Kan-Harada, 2019-08-14:
遭遇したことのないところで落ちてますね・・aravis-0.6は入っているでしょうか。Install.shの中でインストールしているので大丈夫な気はしますが。

コンソールから
   arv-tool-0.6
と入力すると、カメラが接続されていればそれらしくメッセージがでてきます。

Kan-Harada, 2019-08-14:
会ったことのないエラーです。

Pythonのバージョンとか、きちんと合ってますか (2.7)

YukiYamagata, 2019-08-16:
上記のエラーが出た際は、Python3.6にしていたため、
Python2.7.16に変更し、catkin_makeをし直しました。
すると、同じようなエラーがでました。

Traceback (most recent call last):
  File "/home/yuuki/catkin_ws/src/camera_aravis/cfg/CameraAravisConfig.cfg", line 4, in <module>
    import roslib; roslib.load_manifest(PACKAGE)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ImportError: No module named rospkg
camera_aravis/CMakeFiles/camera_aravis_gencfg.dir/build.make:63: ターゲット '/home/yuuki/catkin_ws/devel/include/camera_aravis/CameraAravisConfigConfig.h' のレシピで失敗しました
make[2]: *** [/home/yuuki/catkin_ws/devel/include/camera_aravis/CameraAravisConfigConfig.h] エラー 1
CMakeFiles/Makefile2:3867: ターゲット 'camera_aravis/CMakeFiles/camera_aravis_gencfg.dir/all' のレシピで失敗しました
make[1]: *** [camera_aravis/CMakeFiles/camera_aravis_gencfg.dir/all] エラー 2
make[1]: *** 未完了のジョブを待っています....
[ 46%] Built target rovi_generate_messages_eus
[ 56%] Built target rovi_generate_messages_py
[ 63%] Built target rovi_generate_messages_lisp
Makefile:138: ターゲット 'all' のレシピで失敗しました
make: *** [all] エラー 2
Invoking "make -j4 -l4" failed

YukiYamagata, 2019-08-16:
解決しました。
pyenvを行ったため、ROSをインストールした時のpythonのversionに変更する必要がありました。
$ pyenv global system
を実行することによって解決できました。

closed 2019-08-17 by YukiYamagata