YOODS/rovi #59: rovi実機が認識されない

KoheiYorimoto, 2019-08-14:
README.mdやGettingStarted.mdなどに従いrovi実機とIPアドレスを合わせたのですが，
実機が認識されず，カメラでの撮影ができません．
 
エラー内容
・ カメラ接続後に $ ping 192.168.222.10 を実行 
   →Get value from agent failed: cannot connect to [１92.16.222.10]
・ カメラ接続後に $ arv-tool-0.6を実行
   → No device found
・ カメラ接続後に $ roslaunch rovi ycam3sxga.launch を実行
   →添付画像にて
![Uploading Screenshot from 2019-08-14 16-56-35.png…]

行ったこと
・ パソコンのIPアドレスを192.168.222.10に固定
   →pingは通るようになった
・ GettingStarted.md のaravisの項目にある「エラーになるときの．．．」を実行
    → エラー内容と変わらず
・ /rovi/camera/address を192.168.222.10に設定
    → エラー内容と変わらず

arv-tool-0.6 でカメラが認識されてないのが原因だと思うのですが，認識させるにはどうしたらよいのでしょうか．

KoheiYorimoto, 2019-08-14:
添付画像はこちらです．

[image]

Kan-Harada, 2019-08-14:
192.168.222.10はカメラのアドレスです。
パソコンのアドレスは同じネットワーク内の違うアドレスに設定してください。具体的には
192.168.222.xxx (xxxは1〜255)
です。

KoheiYorimoto, 2019-08-14:
パソコンのIPアドレスを192.168.222.100にするとカメラが認識できました．ありがとうございました．

closed 2020-10-21 by KazukiHiraizumi