YOODS/rovi #63: rovi/GettingStarted.mdの改善案

YukiYamagata, 2019-09-11:
GigEインターフェース設定
●アドレス設定
項目　　　　値　　　                         備考
アドレス　　  192.168.222.100
ネットマスク　255.255.255.0
ゲートウェイ                                          設定不要
MTU             9000                              Jumboパケット対応のため

●設定方法
$ ifconfig
を実行し、インターフェース名(接続名)を取得

$ sudo vi /etc/network/interfaces
を実行し、以下の内容を変更する。
元々の記述はコメントアウトする。

●変更前
auto {取得したインターフェース名}
iface {取得したインターフェース名} inet dhcp

●変更後
auto {取得したインターフェース名}
iface {取得したインターフェース名} inet static
 address 192.168.222.100
 netmask 255.255.255.0
 mtu 9000

※上記の変更前の記述が元々ない場合は、変更後の記述を追記する。

Kan-Harada, 2019-09-11:
/etc/network/interfacesは伝統的なdebianのネットワーク設定ファイルですが、UbuntuやMintのGUIでネットワーク設定を行った場合、/etc/NetworkManager/system-connections/にNIC毎に設定ファイルができるようです。ここを追加・削除するのが正解な気がします。ただ、両方に設定がある場合は/etc/network/interfacesが優先されるような・・・

YukiYamagata, 2019-09-12:
nmcliコマンドでIPアドレス等の設定を行う方法を調べていて、
IPアドレスとサブネットマスクは設定できたのですが、MTUの設定方法がわかりませんでした。
ご教示頂けると幸いです。

YukiYamagata, 2019-09-12:
また、改善案として
RoVIの実行手順のテストにおいて、以下の記述を追記するのはどうでしょうか？
私達が実際に撮像のテストをしてみて、本当に撮像できたのか確認できず不安でしたので、
以下の記述があると確認できると思います。

$ rosservice call /rovi/pshift_genpc
を実行すると、13枚の撮影画像データ(png画像)が保存される。
保存場所は「/imp/」である。capt0*_0.pngという形式で13枚の画像が保存される。
	capt00_0.png:フラッシュと焚かないで撮った画像
	capt01_0.png,:全体をフラッシュ焚いて撮った画像
	capt02_0.png:右半分がフラッシュ焚いて撮って画像
	capt03_0.png:左10分の9ぐらいをフラッシュ焚いて撮った画像
	capt04_0.png:両端3分の1ずつをフラッシュ焚いて撮った画像
	　・・・

test.plyが3次元の点群画像

closed 2020-10-21 by KazukiHiraizumi